raspberrypi/pico-sdk #415: Disable ADC missing

Reneg973, 2021-05-17:
According to datasheet 4.9.3
>The ADC can be disabled again at any time by clearing CS.EN, to save power.

The implemenation for this is missing.

What would be the accepted implementation, #if or #else?

```
static inline void adc_set_enabled(bool enable) {
#if 0
  adc_hw->cs = enable ? adc_hw->cs | ADC_CS_EN_BITS : adc_hw->cs &~ ADC_CS_EN_BITS;
#else
  enable
    ? hw_set_bits(&adc_hw->cs, ADC_CS_EN_BITS)
    : hw_clear_bits(&adc_hw->cs, ADC_CS_EN_BITS);
#endif
}
```

bruelltuete, 2022-04-24:
I would call it `adc_deinit()`, similar to `i2c_deinit()`.

Reneg973, 2022-04-26:
> 

> I would call it `adc_deinit()`, similar to `i2c_deinit()`.

Not really, I don't expect a deinit function to have a parameter. This function really can enable or disable the ADC, so it doesn't fit

Reneg973, 2022-04-26:
> > > I would call it `adc_deinit()`, similar to `i2c_deinit()`.
> > 
> > 
> > Not really, I don't expect a deinit function to have a parameter. This function really can enable or disable the ADC, so it doesn't fit
> 
> Do you mean you don't like `adc_deinit(bool enable)`? Yeah me neither... it would just be `adc_deinit()` without param.

No I think you are out of context. The point is with setting/resetting the flag you enable or disable the ADC conversion. It doesn't do any deinitialization. You can simply enable it by setting the bit again. The same applies for gpio_deinit(). There is no deinit on a GPIO.