raspberrypi/pico-sdk #420: spi.c explicitly removes __not_in_flash_func(), but no reason why.

smithps, 2021-05-19:
I have been looking to speed up some code of mine which handles a display running over SPI. Delving into spi.c I see that there is the following code, but without an explanation/comment.

`
#pragma warning "spi.c: removing __not_in_flash_func() directive"
#undef __not_in_flash_func
#define __not_in_flash_func(x) x

`

I also see that as well as some functions which are declared _inline_, and some with no decorators, that some functions _are_ declared as __not_in_flash_func() .... so someone must have decided that these should be in flash for a reason, and rather than removing the directive in the function declaration, that has been left there and the define has been undef'ed.

I am assuming that me putting them back into RAM isn't going to break anything, but being rather anal about such things it would be nice if there was a comment in the code to explain why these functions are in flash, even though they have declarations that state they are  __not_in_flash_func().

lurch, 2021-05-19:
Where are you looking? I can't see that in https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_spi/spi.c

smithps, 2021-05-19:
> Where are you looking?

Hi Lurch,
    seems like I am the culprit. I just redownloaded the SDK and diffed it against the one I had and it appears that that was the only change.

Apologies.

smithps, 2021-05-19:
> > Where are you looking?
> 
> Hi Lurch,
> seems like I am the culprit. I just redownloaded the SDK and diffed it against the one I had and it appears that that was the only change.
> 
> Apologies.

What's bad is that I don't even recall doing it!

closed 2021-05-19 by smithps