raspberrypi/pico-sdk #427: ph_create might dereference a null pointer

jonathangjertsen, 2021-05-21:
`ph_create` has this code:

```C
    pheap_t *heap = calloc(1, sizeof(pheap_t));
    heap->nodes = calloc(max_nodes, sizeof(pheap_node_t));
```

This causes a null pointer dereference if the first `calloc` fails.

pheap.h also has a similar problem as issue #426 - the documentation should mention that the returned pointer might be null, and that the user should check this

Reneg973, 2021-05-21:
and what would be the error handling if the check fails?

jonathangjertsen, 2021-05-21:
For example:

```C
    pheap_t *heap = calloc(1, sizeof(pheap_t));
    if (heap == NULL) {
        return NULL;
    }    
    heap->nodes = calloc(max_nodes, sizeof(pheap_node_t));
    if (heap->nodes == NULL) {
        free(heap);
        return NULL;
    }
    ph_post_alloc_init(heap, max_nodes, comparator, user_data);
    return heap;
```

The user can handle the case where the return value is NULL in a way that is suited to their application

Reneg973, 2021-05-22:
I assume you didn't follow my mind :)
The first question is, who is the user? If you now answer "the developer", you are right. For me (as a developer) it is much better, when I see that during testing a hard fault appears, than to write error handling code, which should have been done much before I call this function in case the RAM is so full that it is not able to allocate a small `pheap_t` struct.
In reality it means, the code has to be refactored so, that this situation should not appear at all.

My conclusion is, better to have a small code base that hard faults, than to have a lot of error handling code, which is not called in 99% of the cases, because it is just an error, and we - as developers - should try to code happy day scenarios instead of hacking until nothing goes anymore and then handle the appearing errors.
Am I alone with this approach?

jonathangjertsen, 2021-05-22:
Thanks for the explanation, I think I agree. I was about to suggest adding asserts, but I see that this is actually already done inside the calloc implementation.

closed 2021-05-22 by jonathangjertsen