raspberrypi/pico-sdk #429: initializing sdk in subdirectory ignores the cross-toolchain

nonchip, 2021-05-21:
```cmake
# /CMakeLists.txt
cmake_minimum_required(VERSION 3.13)
project(project)
add_subdirectory(host_application) # where host stuff will happen
add_subdirectory(firmware)
```
```cmake
# /project/CMakeLists.txt
include(pico_sdk_import.cmake)
project(project-firmware C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
pico_sdk_init()
add_executable(xip
  xip.c
)
target_link_libraries(xip
  pico_stdlib
  pico_bootrom
)
pico_add_extra_outputs(xip)
pico_set_binary_type(xip default)
```

when i do (inside a `/build` folder) `cmake ..` i get:
```
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
[...]
Using PICO_SDK_PATH from environment ('/home/kyra/src/pi_pico/pico-sdk')
PICO_SDK_PATH is /home/kyra/src/pi_pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
[...]
```
and consequently lots of fun errors like:
```
$ make       
Scanning dependencies of target bs2_default
[  1%] Building ASM object firmware/pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.o
/home/kyra/src/pi_pico/pico-sdk/src/rp2_common/boot_stage2/boot2_w25q080.S: Assembler messages:
/home/kyra/src/pi_pico/pico-sdk/src/rp2_common/boot_stage2/boot2_w25q080.S:89: Error: unknown pseudo-op: `.syntax'
```

when i do `cmake ../firmware` i get:
```
Using PICO_SDK_PATH from environment ('/home/kyra/src/pi_pico/pico-sdk')
[...]
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
[...]
```
and it builds.

looks to me like `pico_sdk_init` fails to set the ARM compiler if **any other project** has already found a (non-cross) compiler.
even without the other project/subdirectory being included, the fact the top level list always *per definition* contains a `project` statement (which, per default, initialize the `C CXX` languages even if they don't even contain a target) fails this.

nonchip, 2021-05-21:
note especially the line `PICO_GCC_TRIPLE defaulted to arm-none-eabi` missing when included as a subdirectory, hinting at `cmake/preload/toolchains/pico_arm_gcc.cmake` never actually being loaded, despite the `Defaulting PICO platform compiler to pico_arm_gcc since not specified.` from `cmake/preload/platforms/pico/pico.cmake` sounding very much like it should.

kilograham, 2021-05-21:
Yes, you cannot mix multiple compilers in the same build... you need to use something like ExternalProject_Add if you want to cobble together multiple builds for different systems

kilograham, 2021-05-21:
When i have host stuff and not stuff; i may use the same CMake project but with different build directories (and IFs in the cmakelists.txt to decide what to build)

closed 2021-05-21 by kilograham