raspberrypi/pico-sdk #435: Some code in the "common" libraries is incompatible with MSVC

jonathangjertsen, 2021-05-22:
I encountered some issues while trying to build parts of the pico sdk for MSVC (for host-based unit testing):

* using the `invalid_params_if`, `valid_params_if`, `hard_assert_if` macros triggers errors like `[...]\pico-sdk\src\host\hardware_timer\include\hardware/timer.h(22): error C2059: syntax error: '{'`
* `auto_init_mutex` uses `__attribute__` which is a GCC extension
* `sem_available` uses `typeof` which is a GCC extension
* `hw_divider_result_threadlocal` is declared with `__thread`, but `platform.h` (which defines `__thread`) is not included by the header file

I have workarounds for these issues in https://github.com/jonathangjertsen/pico-sdk/commit/0fa7da92dffa83500faba3c32faaff1de2049efd, but they are hacks and only cover the parts I used in my application. A quick search through the code base reveals many other usages of GCC extensions under the "common" code.

To what extent is building for host with MSVC supported? It seems like at least some support was intended at some point, since there are a couple of checks for `_MSC_VER`.

kilograham, 2021-05-23:
Yeah it isn't really supported atm as (as you saw) started down that route, but there are a bunch of (as you say non-trivial) issues; hoping that gcc is a reasonable workaround on WIndows, but happy to accept PRs to fix either, we just don't have any resources to spend on it atm.

ajf58, 2025-01-19:
This is still an issue with v2.1.0.