raspberrypi/pico-sdk #440: Host implementation of pico_multicore

jonathangjertsen, 2021-05-23:
It would be nice to have a host implementation of the multicore functionality which uses threads (or processes?) in place of the two cores.

I've started on an implementation here: https://gist.github.com/jonathangjertsen/590141bc17ffb47d0f091c6ab82396f3
The test uses the Unity framework. I haven't made a PR because there is a known bug (`test_multicore_fifo_pop_timeout_us` fails half the time and I don't know why), and also because I couldn't figure out how to integrate it into the pico-sdk build system. Feel free to build on it though.

kilograham, 2021-05-24:
there is an implementation in `pico_host_sdl` that uses SDL

kilograham, 2021-05-24:
https://github.com/raspberrypi/pico-host-sdl/blob/master/README.md

jonathangjertsen, 2021-05-24:
Yes I saw, but it is incomplete (in particular, no FIFO support) and it would be nice to avoid having to pull in SDL if you're testing something other than audio/video.

kilograham, 2021-05-24:
Yeah, i would probably have included threads by default were pthread available on windows; that said building with MSVC has issues anyway; not sure if there is a pthread impl for windows for use with gcc (or other)

> (in particular, no FIFO support)

Frankly, we are discouraging the use of the FIFOs (for most things anyway) anyway, as they are a unique resource  (needed by RTOS impls, core 1 launch, and interrupting the other core)...

jonathangjertsen, 2021-05-24:
> not sure if there is a pthread impl for windows for use with gcc (or other)

MinGW has it - not sure if it's included by default or if it has to be installed