raspberrypi/pico-sdk #442: host_hid mouse data displaced

dshadoff, 2021-05-24:
Some wired mice (such as the official Raspberry Pi mouse) work with the host_hid USB example code in pico-examples - in fact, it seems that at the moment, only wired mice are recognized (although this is not the issue I wish to report here).

At least some wired mice which have side buttons (i.e. back/forward) have displaced data:
- button presses appear in the (hid_mouse_report_t).x element rather than (hid_mouse_report_t).button;
- x movement appears in (hid_mouse_report_t).y rather than (hid_mouse_report_t).x .
- (hid_mouse_report_t).button remains 0

(note: the mouse works fine on a RPi 400)

I am reporting this here as pico-sdk holds a non-current version of tinyusb, and it doesn't seem to be straightforward to test with the current tinyusb on pico hardware at the moment.

This may be solved with the newer version of tinyusb in pico-sdk 1.2.0 ... this issue can serve as a placeholder until that is available for test; if pico-sdk and tinyusb are in sync at that time (and if the issue is not yet resolved), I will be happy to report the issue upstream.  I am also happy to gather any diagnostic information required (if the process is explained).

For the moment, the existence of a report may be helpful to other users, to indicate current known system limitations.

kilograham, 2021-05-31:
please try with tinyusb-0.10.0 branch of this repo

dhalbert, 2021-05-31:
Tagging @hathach

dshadoff, 2021-06-01:
Having some trouble actually testing this.

I tried this by re-cloning the pico-sdk branch tinyusb-0.10.0 (in the original pico-sdk location), followed by rebuilding pico-examples.

When I first ran "cmake -B build" in pico-examples, a CMake warning came back, requiring me to go to the pico-sdk repo and run "git submodule update --init", in order to build any tinyusb examples (which I did).

I am assuming that this checked out the correct version of raspberrypi/tinyusb into pico-sdk/lib/tinyusb, although I wasn't able to find the commit ID on the expected branches.  (commit ID was a long number terminating in "934cd935")

The build failed during the build of the tinyusb sources, in usb/device/dev_audio_headset, so I'm wondering if perhaps the link in the pico-sdk/lib/tinyusb on the tinyusb-0.10.0 branch is not fetching an appropriate version of the tinyusb module...

kilograham, 2021-06-01:
ah - yeah, there isn't a checked in version o pico-examples to match tinyusb-0.10.0 (it has changed though)

you should be able to build from the examples/device directory in tinyusb though with PICO_SDK_PATH set.

dshadoff, 2021-06-01:
To be clear, do you mean to:
`cd (base)/pico-sdk/lib/tinyusb/examples/device`
and then
```
cmake -B build
cd build
make
```

...Because this had problems during the cmake step:
```
CMake Error at cdc_msc_freertos/CMakeLists.txt:19 (message):
  Invalid FAMILY specified:

-- Configuring incomplete, errors occurred!

```

kilograham, 2021-06-01:
that's a weird error (that makes me think you don't have the latest pico-0.10.0 tinyusb code)

That said, you need to specify a FAMILY and a BOARD e.g.

kilograham, 2021-06-01:
`cmake -DFAMILY=rp2040 -DBOARD=raspberry_pi_pico' -B build`

you can use `-DBOARD=pico_sdk` too to just follow PICO_BOARD setting

dshadoff, 2021-06-01:
That failed in a different way.

As for the version of the pico-0.10.0 tinyusb code, I cloned the tinyusb-0.10.0 branch of pico-sdk last night, and ran "git submodule update --init" which brought in whatever version was linked from lib/tinyusb in that branch... but I too was unclear whether it was the correct one.

Here is the most recent output:

```
Using PICO_SDK_PATH from environment ('/home/pi/devel/pico/pico-setup/pico/pico-sdk')
PICO_SDK_PATH is /home/pi/devel/pico/pico-setup/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
PICO target board is pico.
Using board configuration from /home/pi/devel/pico/pico-setup/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.7 (found version "3.7.3") found components: Interpreter 
CMake Warning at ../../../../src/rp2_common/tinyusb/CMakeLists.txt:14 (message):
  PICO_TINYUSB_PATH specified but content not present.

Using PICO_EXAMPLES_PATH from environment ('/home/pi/devel/pico/pico-setup/pico/pico-examples')
-- Configuring done
CMake Error at net_lwip_webserver/CMakeLists.txt:16 (add_executable):
  Cannot find source file:

    /home/pi/devel/pico/pico-setup/pico/pico-sdk/lib/tinyusb/lib/lwip/src/core/altcp.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

```

dshadoff, 2021-06-01:
Actually I realized...
The example I was looking to use is found in host, not device (host hid mouse device, as per the issue title).
So building in examples/device would appear to be incorrect.

When building in examples/host, it also failed (but in a different way):
cmake was fine
But make failed:
```
Scanning dependencies of target raspberry_pi_pico-cdc_msc_hid
[ 20%] Building C object cdc_msc_hid/CMakeFiles/raspberry_pi_pico-cdc_msc_hid.dir/src/hid_app.c.obj
/home/pi/devel/pico/pico-setup/pico/pico-sdk/lib/tinyusb/examples/host/cdc_msc_hid/src/hid_app.c:26:10: fatal error: bsp/board.h: No such file or directory
 #include "bsp/board.h"
          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [cdc_msc_hid/CMakeFiles/raspberry_pi_pico-cdc_msc_hid.dir/build.make:63: cdc_msc_hid/CMakeFiles/raspberry_pi_pico-cdc_msc_hid.dir/src/hid_app.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1631: cdc_msc_hid/CMakeFiles/raspberry_pi_pico-cdc_msc_hid.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```

kilograham, 2021-06-02:
I don't think i realized you were building the standalone "tinyusb way" from within the tinyusb submodule of pico-sdk.
I assumed you were building from a standalone tinyusb depo pointing at a standalone pico-sdk via PICO_SDK_PATH

in any case, there pico-sdk->tinyusb did not have the latest submodule commit reference; i have updated

hathach, 2021-06-02:
@dshadoff can you clone the upstream and cd to  https://github.com/hathach/tinyusb/tree/master/examples/host/cdc_msc_hid then

```
make BOARD=raspberry_pi_pico all
```

then copy the uf2 to pico to test with.

dshadoff, 2021-06-02:
OK, I interpreted that as:
```
cd (parent of pico-sdk)
git clone git://github.com/hathach/tinyusb
cd tinyusb/examples/host
make BOARD=raspberry_pi_pico all
```

This creates a separate tinyusb repository, adjacent to pico-sdk, and on the master git branch (which is more current than 0.10.0, as it had a commit as recently as 9 hours ago).  Also, my pico-sdk is still on the tinyusb-0.10.0 branch.

Results are mostly good news:

The build went fine
The execution does not crash as my previous attempt did.
1) Raspberry Pi official mouse works as well as the previous pico-sdk examples did.
2) Logitech wireless mouse is still not recognized at all.
3) The wired mouse with forward/back side buttons works better than the previous release, but still not completely correctly:
- all 4 buttons now are recognized correctly and do not cause any confusing 'X' or 'Y' movements
- 'X' movements appear fine, except that the 'Y' value is either "0" (for positive X movement) or "-1" (for negative X movement)
- 'Y' movement is not reported in the report->Y, but rather in the wheel value
So for some reason, this mouse appears as though it may be reporting 16-bit X and Y values ?  In any case, it works fine on Windows and Linux machines...

hathach, 2021-06-02:
sure there is a bug, now let test with the logitech mouse
1. can you uoload its report descriptor as txt file. If you are in linux you could dump it with `sudo usbhid-dump -d cafe | grep -v : | xxd -r -p | hidrd-convert -o spec` replace `cafe` with logitech vid
2. now run compile again with log enable (need to clean first)
```
make BOARD=raspberry_pi_pico clean
make BOARD=raspberry_pi_pico LOG=3 all
```

hook and ftdi/cp21xx or pico probe to uart of the pico, then post your log as txt file for analyzing. 

you could also post the log with the 3) wired mouse fwd/bck not work right, but that would be later or better a separated issue.

dshadoff, 2021-06-02:
What do you mean "logitech vid" ?  How would I find that ?

dshadoff, 2021-06-02:
Here is the capture with Logitech unifying receiver.
The HID example initializes, but does not react to any mouse movement or clicking (with or without logging on), but there was some useful-looking log output of the initialization.

[logitech_pico_log.txt](https://github.com/raspberrypi/pico-sdk/files/6582070/logitech_pico_log.txt)

hathach, 2021-06-02:
@dshadoff I will need both the configuration descriptor from `lsusb -v` and report descriptor which hdrd-convert command  as well.

```
sudo usbhid-dump -d cafe | grep -v : | xxd -r -p | hidrd-convert -o spec
```

dshadoff, 2021-06-02:
OK, lsusb provides the following:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04d9:0007 Holtek Semiconductor, Inc. 
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
From the above, I am assuming that instead of "cafe", I should be using the "046d" portion of the Logitech device ID ?
The output from that is in this file:  
[lsusb-Logitech.txt](https://github.com/raspberrypi/pico-sdk/files/6584268/lsusb-Logitech.txt)

Running the final command I run into a problem, as "hidrd-convert" is not found on the Pi 400 I am using; where can I obtain it ?

dshadoff, 2021-06-02:
Never mind,  was able to locate it and get it with:
`sudo apt-get install hidrd usbutils`

The output is attached here:
[hidrd-Logitech.txt](https://github.com/raspberrypi/pico-sdk/files/6584387/hidrd-Logitech.txt)

v9938, 2021-06-03:
Hi hathach

I have the same problem. 
tinyusb-0.10.1 didn't fix it.

I used the USB packet analyzer.
[image]

It seems that GET_DESCRIPTOR(DEVICE) did not receive the requested size.
I think it misjudged the length of the IN packet of Raspberry Pi Pico.

If there is anything we can do to help, please request it.

dshadoff, 2021-06-03:
@v9938 To clarify, when you say "the same problem", are you referring to:
a) The original problem, where a wired mouse with side buttons has offset data in X/Y reports ?  or
b) The issue where a wireless mouse (in my case, Logitech) does not process data at all (which @hathach asked for more information, roughly 36 hours ago) ?
These should probably be separate issues (but now that we're here, I'm not clear on how to split/splice them).
My original thought had been to wait for pico-sdk v1.2.0 to be released, and then go upstream to tinyusb, once they were roughly in sync.

v9938, 2021-06-03:
Hi dshadoff
My answer ’B’.
I am trying to get this problem solved with pico-sdk branch tinyusb-0.10.1. 

If it is inappropriate, I can discuss it in another topic.

dshadoff, 2021-06-03:
@kilograham and @hathach should decide how to proceed.  I just wanted to clarify your comment.

hathach, 2021-06-04:
@dshadoff thanks for the info, I am also able to reproduce the issue with one of my laying dongle and spent a couple hours troubleshooting it. It needs a bit of works, here is what I found so far.
- logitech dongle use 8 bytes control endpoint in **Full speed** mode, which is different than 8 bytes control endpoint in Low speed mode fixed by https://github.com/hathach/tinyusb/pull/862
- some of https://github.com/hathach/tinyusb/pull/862 need to be reverted, since submitting  more than maxpacket_size per buffer ( 64 instead of 8) does not work with FS device (although it work with LS)
-  RP2040-E5 walkaround need more fix: e.g https://github.com/hathach/tinyusb/blob/master/src/portable/raspberrypi/rp2040/rp2040_usb.c#L211 transferred_bytes should be extracted after buf_sel adjust
- RP2040-E5 walkround somehow still cause an additional IN token which the dongle responded with STALL, if it is just NAK we would be fine. But we cannot blame it since we send more IN than needed

Overall, since I want to implement double buffering at least for host in the near future, I feel that keep fixing the RP2040-E5 walkaround doesn't justify the effort. It is best to use implement double buffer to get in sync with hardware. Though I need a bit more time to separate the hcd and dcd endpoint transfer, since it is easy to mess up the device side just now. Better to do it in incremental step. I will update this topic whenever there is a new PRs for this.

[image]

lurch, 2021-06-04:
Is https://github.com/raspberrypi/tinyusb/pull/7 relevant here @hathach ? Or should that PR be closed already?

hathach, 2021-06-07:
> Is [raspberrypi/tinyusb#7](https://github.com/raspberrypi/tinyusb/pull/7) relevant here @hathach ? Or should that PR be closed already?

You should ask the OP of that issue. The host is current has some bugs related to the E4 anyway, I will try to see if I could get it all resolved soon enough.

dshadoff, 2021-06-09:
I know that you're looking at the wireless Logitech logs (and I hope progress is going well), but now I have also captured the details for the original wired 4-button mouse:

lsusb says:
Bus 001 Device 005: ID 18f8:1286 [Maxxter]

The hidrd output for teh device is here:
[wired_4button_hidrd.txt](https://github.com/raspberrypi/pico-sdk/files/6621104/wired_4button_hidrd.txt)

And the output from the special (LOG=3) build of the host example is here:
(For movements, I moved left, then right, then up, then down, then pressed left button, right button, back button, and forward button).
[wired_4button_pico_log.txt](https://github.com/raspberrypi/pico-sdk/files/6621105/wired_4button_pico_log.txt)

hathach, 2021-06-09:
Thanks any data available in advance is good when I switch back to work on tinyusb (on other work just now).

hathach, 2021-06-11:
@dshadoff https://github.com/hathach/tinyusb/pull/891 should fix issue with your logitech receiver, please check it out to see if that works for you. It is still WIP, and need more testing.

v9938, 2021-06-11:
@hathach Some of the problems were solved.
But a new problem occurred. HID open fails.
I have attached the [debug.log](https://github.com/raspberrypi/pico-sdk/files/6640368/debug.log)

Please let me know if there is any other information needed.

dshadoff, 2021-06-11:
@hathach  thank you for your progress !
I git-cloned the host-rp2040-double-buffer branch of tinyusb and built from that (it includes a few more updates than just PR 891).

A previously non-working wireless mouse ( [https://www.adafruit.com/product/1738](https://www.adafruit.com/product/1738) ) is now working, however as @v9938 stated, the Logitech one is still not yet working.  Logs attached.
[logitech_pico_log_20210611.txt](https://github.com/raspberrypi/pico-sdk/files/6640918/logitech_pico_log_20210611.txt)

P.S. no change to the wired mouse with side-buttons, but I expect this is not yet a focus.

hathach, 2021-06-12:
@dshadoff @v9938 ah this one is simple, the logitech dongle has 3 HID interfaces, the example only configure to work up to 2. I have increased CFG_TUH_HID = 4 in example. Please try again.

@dshadoff mouse report format can be different from mouse, the example only parse using the boot mouse format. Which is common used enough. If it  does not match, you should look at its report descriptor (hidrd), test its report sent to PC and parser accordingly. i.e hid_mouse_report_t is not always correctly mapped to all mouse report.

v9938, 2021-06-12:
@hathach @dshadoff 
Thanks for the nice support.
I have tried changing the sample to 'CFG_TUH_HID = 4'.

I found that there is still one problem.
1. HID_Open problem fixed.
2. 1st-time connection has no problem.
3. After the second time connection, USBH DEVICE ATTACH and REMOVE are repeated.

The debug log as follows.
[logitech_pico_log_20210612.txt](https://github.com/raspberrypi/pico-sdk/files/6642374/logitech_pico_log_20210612.txt)

The test scenario followed the steps below.
1. Connect Logitech receiver.
2. Power ON [LOG L1-]
3. Type 123456790 on the keyboard [LOG L743-]
4. Press and release the left mouse button.[LOG L1023-]
5. Press and release the right mouse button.[LOG L1053-]
6. Disconnect Logitech receiver [LOG  L1083-]
7. Connect Logitech receiver [LOG L1092-]

hathach, 2021-06-12:
try to run with LOG=1, or 0, too much printf can somehow cause some issue. If it still exists, it is entirely new bug, and should be filed as separated issue.

dshadoff, 2021-06-12:
I have tried the new version and it is now reacting to the Logitech mouse, but there are two items:
1) In the same way as my earlier report about the wired mouse with side buttons, the X/Y data is offset (although it is different here).  Log level 1 of this :  
[logitech_pico_log_20210612.txt](https://github.com/raspberrypi/pico-sdk/files/6642681/logitech_pico_log_20210612.txt)

> mouse report format can be different from mouse, the example only parse using the boot mouse format. Which is common used enough. If it does not match, you should look at its report descriptor [...]

I basically understand what you are asking me, but it seems I need deeper knowledge of how the library works.  The example is certainly not showing how to do this.  Where can I look for more information ?

2) as @v9938 mentioned, removing the receiver and reconnecting it, it recognizes the receiver but does not reconnect to the mouse.  Log level 1 of this:   
[logitech_pico_log_20210612_disconnect_reconnect.txt](https://github.com/raspberrypi/pico-sdk/files/6642686/logitech_pico_log_20210612_disconnect_reconnect.txt)

dshadoff, 2021-06-12:
Looking at the reports at the LOG=1 level, I have determined the following:

Simple wired mouse (like RPi standard mouse):
byte 0 = buttons (8-bit, bitmask)
byte 1 = X (8-bit value, signed)
byte 2 = Y (8-bit value, signed)
byte 3 = scroll (8-bit value, signed)
bytes 4-7  = apparently unused

My example Wired mouse with side buttons:
byte 0 = literal value of 01 hexadecimal
byte 1 = buttons (8-bit, bitmask)
byte 2 / 3 = X (16-bit value, LSB followed by MSB)
byte 4 / 5 = Y (16-bit value, LSB followed by MSB)
byte 6 = scroll (8-bit value, signed)
byte 7  = apparently unused

Logitech mouse:
byte 0 = literal value of 02 hexadecimal
byte 1 = buttons (8-bit, bitmask)
byte 2 = apparently unused (or perhaps unused buttons ?)
byte 3 = X (LSB of 12-bit value)
byte 4 = Y least-significant nybble + X most-significant nybble  (i.e. 0x1F: Y = 1, X is negative)
byte 5= Y (MSB of 12-bit value)
byte 6 = vertical scroll (8-bit value, signed)
byte 7 = horizontal scroll (8-bit value, signed)
** The 12-bit encoding is hard to explain, but  if X=-3 and Y = 3, this would show as bytes 3/4/5 = FD 3F 00

But I would like to know:
a) how to differentiate between which one to parse
b) where to obtain these values as raw data so that I can map them properly
c) why the mouse report uses only 8-bit variables for X/Y if reported values from the mouse can be larger... or are there perhaps multiple mouser report structures I should be aware of ?

v9938, 2021-06-12:
@hathach @dshadoff 
I created #483 about the connection problem with Logitech USB receiver. 
Please comment if there are any problems.

hathach, 2021-06-13:
> I basically understand what you are asking me, but it seems I need deeper knowledge of how the library works. The example is certainly not showing how to do this. Where can I look for more information ?

The report representation format (which byte for which usage) is detailed in report descriptor. HID parser is very complicated (most complicated descriptor of USB), and TinyUSB basically cannot parse it all. It currently only provide a very simple parser to parse usage_page and usage of each report ID. For more than that, application have to parse it themself, the report descriptor is available with the `tuh_hid_mount_cb()`

For mouse and keyboard, we can actually use boot interface which guarantee to work on all mouse/keyboard. For boot support, I will do it later on when having time. If you want to understand more about HID, check out the USB HID specs, it is rather short and simple. 

For now, I will wrap up my work here with the double buffered. Others are unrelated issue, and should be addressed separately.

hathach, 2021-06-17:
@dshadoff I made another branch to force/set boot protocol for keyboard, mouse device here https://github.com/hathach/tinyusb/tree/more-hid-host (since double buffered branch is under review). This should solve your issue with mis-matched report with your logitech/wired mouse.

v9938, 2021-06-17:
@dshadoff  I immediately tested it.
It was an improvement over the previous version.
It was a case of three things.

1. PANIC occurred
```
....
A device with address 1 is unmounted
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 1 is unmounted
HID device address = 1, instance = 2 is unmounted

*** PANIC ***

ep 0 in was already available
```

2. RESET? occurred
```
....
A device with address 1 is mounted
----------------------------------------------------
TinyUSB Host Example
If you find any bugs or problems, feel free to open
an issue at https://github.com/hathach/tinyusb
----------------------------------------------------

This Host demo is configured to support:
  - RTOS = Raspberry Pi Pico SDK
----------------------------------------------------
TinyUSB Host Example
If you find any bugs or problems, feel free to open
an issue at https://github.com/hathach/tinyusb
----------------------------------------------------

This Host demo is configured to support:
  - RTOS = Raspberry Pi Pico SDK
----------------------------------------------------
TinyUSB Host Example
If you find any bugs or problems, feel free to open
an issue at https://github.com/hathach/tinyusb
----------------------------------------------------

This Host demo is configured to support:
  - RTOS = Raspberry Pi Pico SDK
HID device address = 1, instance = 0 is mounted
HID has 1 reports and interface protocol = Keyboard
HID device address = 1, instance = 1 is mounted
HID has 4 reports and interface protocol = Mouse
HID device address = 1, instance = 2 is mounted
HID device address = 1, instance = 0 is mounted
```

3. usbh_edpt_xfer 386: ASSERT FAILED  occurred
```
....
  - RTOS = Raspberry Pi Pico SDK
HID device address = 1, instance = 0 is mounted
HID has 1 reports and interface protocol = Keyboard
HID device address = 1, instance = 1 is mounted
HID has 4 reports and interface protocol = Mouse
HID device address = 1, instance = 2 is mounted
HID has 3 reports and interface protocol = None
A device with address 1 is mounted
A device with address 1 is unmounted
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 1 is unmounted
HID device address = 1, instance = 2 is unmounted
HID device address = 1, instance = 0 is mounted
HID has 1 reports and interface protocol = Keyboard
usbh_edpt_xfer 386: ASSERT FAILED
HID device address = 1, instance = 1 is mounted
HID has 4 reports and interface protocol = Mouse
usbh_edpt_xfer 386: ASSERT FAILED
HID device address = 1, instance = 2 is mounted
HID has 3 reports and interface protocol = None
usbh_edpt_xfer 386: ASSERT FAILED
A device with address 1 is mounted
A device with address 1 is unmounted
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 1 is unmounted
HID device address = 1, instance = 2 is unmounted
```

hathach, 2021-06-17:
@v9938 can you provide the context for each case, which device you use to test, and what you have done to reproduce the issue and how often it happens. 1 and 2 loom like you have unstable power or usb connection

v9938, 2021-06-17:
I re-tested.
It seems to depend on the version of the Logicool Unifiy receiver.

**Case1 PANIC occurred**
  Firmware version 024.007.00030
[image]

**Case3: usbh_edpt_xfer 386: ASSERT FAILED occurred**
Firmware version 012.009.00030
[image]
Firmware version 012.010.00032
[image]

**More information...**
When each receiver was connected to a Windows PC, the USB packets were as follows
The max size of EP0 for GET_DESCRIPTOR(device) differs between Ver012 and Ver024.

Firmware version 012.010.00032
```
SOF=049 - 065
SOF=066  SETUP ADR=0   ENP=0 DATA0 80 06 00 01 00 00 40 00   ACK  GET_DESCRIPTOR
                                                                  (device)
SOF=067   I N  ADR=0   ENP=0 DATA1 12 01 00 02 00 00 00 08   ACK  [ ........ ]
          OUT  ADR=0   ENP=0 DATA1                           ACK
SOF=073 - 08D
SOF=08E  SETUP ADR=0   ENP=0 DATA0 00 05 02 00 00 00 00 00   ACK  SET_ADDRESS
                                                                  addr = 2
          I N  ADR=0   ENP=0 DATA1                           ACK
SOF=08F - 0B8
SOF=0B9  SETUP ADR=2   ENP=0 DATA0 80 06 00 01 00 00 12 00   ACK  GET_DESCRIPTOR
                                                                  (device)
SOF=0BA   I N  ADR=2   ENP=0 DATA1 12 01 00 02 00 00 00 08   ACK  [ ........ ]
          I N  ADR=2   ENP=0 DATA0 6D 04 2B C5 09 12 01 02   ACK  [ m.+..... ]
          I N  ADR=2   ENP=0 DATA1 00 01                     ACK  [ .. ]
          OUT  ADR=2   ENP=0 DATA1                           ACK
         SETUP ADR=2   ENP=0 DATA0 80 06 00 02 00 00 FF 00   ACK  GET_DESCRIPTOR
                                                                  (configuration)
          I N  ADR=2   ENP=0 DATA1 09 02 54 00 03 01 04 A0   ACK  [ ..T..... ]
          I N  ADR=2   ENP=0 DATA0 31 09 04 00 00 01 03 01   ACK  [ 1....... ]
          I N  ADR=2   ENP=0 DATA1 01 00 09 21 11 01 00 01   ACK  [ ...!.... ]
          I N  ADR=2   ENP=0 DATA0 22 3B 00 07 05 81 03 08   ACK  [ ";...... ]
SOF=0BB   I N  ADR=2   ENP=0 DATA1 00 08 09 04 01 00 01 03   ACK  [ ........ ]
          I N  ADR=2   ENP=0 DATA0 01 02 00 09 21 11 01 00   ACK  [ ....!... ]
          I N  ADR=2   ENP=0 DATA1 01 22 94 00 07 05 82 03   ACK  [ ."...... ]
          I N  ADR=2   ENP=0 DATA0 08 00 02 09 04 02 00 01   ACK  [ ........ ]
          I N  ADR=2   ENP=0 DATA1 03 00 00 00 09 21 11 01   ACK  [ .....!.. ]
          I N  ADR=2   ENP=0 DATA0 00 01 22 5D 00 07 05 83   ACK  [ .."].... ]
          I N  ADR=2   ENP=0 DATA1 03 20 00 02               ACK  [ . .. ]
          OUT  ADR=2   ENP=0 DATA1                           ACK
         SETUP ADR=2   ENP=0 DATA0 80 06 00 03 00 00 FF 00   ACK  GET_DESCRIPTOR
                                                                  (string)
SOF=0BC   I N  ADR=2   ENP=0 DATA1 04 03 09 04               ACK  [ .... ]
          OUT  ADR=2   ENP=0 DATA1                           ACK
         SETUP ADR=2   ENP=0 DATA0 80 06 02 03 09 04 FF 00   ACK  GET_DESCRIPTOR
                                                                  (string)
          I N  ADR=2   ENP=0 DATA1 1A 03 55 00 53 00 42 00   ACK  [ ..U.S.B. ]
          I N  ADR=2   ENP=0 DATA0 20 00 52 00 65 00 63 00   ACK  [  .R.e.c. ]
          I N  ADR=2   ENP=0 DATA1 65 00 69 00 76 00 65 00   ACK  [ e.i.v.e. ]
          I N  ADR=2   ENP=0 DATA0 72 00                     ACK  [ r. ]
          OUT  ADR=2   ENP=0 DATA1                           ACK
SOF=0BD - 0BE
SOF=0BF  SETUP ADR=2   ENP=0 DATA0 80 06 00 01 00 00 12 00   ACK  GET_DESCRIPTOR
                                                                  (device)
          I N  ADR=2   ENP=0 DATA1 12 01 00 02 00 00 00 08   ACK  [ ........ ]
          I N  ADR=2   ENP=0 DATA0 6D 04 2B C5 09 12 01 02   ACK  [ m.+..... ]
          I N  ADR=2   ENP=0 DATA1 00 01                     ACK  [ .. ]
          OUT  ADR=2   ENP=0 DATA1                           ACK
         SETUP ADR=2   ENP=0 DATA0 80 06 00 02 00 00 09 00   ACK  GET_DESCRIPTOR
                                                                  (configuration)
          I N  ADR=2   ENP=0 DATA1 09 02 54 00 03 01 04 A0   ACK  [ ..T..... ]
[... 256 lines cut ...]
                                   81 03 C0 06 BC FF 09 88        [ ........ ]
SOF=73C   I N  ADR=2   ENP=0 DATA1 A1 01 85 08 19 01 29 FF   ACK  [ ......). ]
                                   15 01 26 FF 00 75 08 95        [ ..&..u.. ]
                                   01 81 00 C0                    [ .... ]
          OUT  ADR=2   ENP=0 DATA1                           ACK
SOF=73D - 744
SOF=745  SETUP ADR=2   ENP=0 DATA0 21 0A 00 00 02 00 00 00   ACK  (Class-specific request)
          I N  ADR=2   ENP=0                                STALL
SOF=746  SETUP ADR=2   ENP=0 DATA0 81 06 00 22 02 00 A2 00   ACK  GET_DESCRIPTOR
                                                                  (CS_CONFIGURATION)
          I N  ADR=2   ENP=0 DATA1 06 00 FF 09 01 A1 01 85   ACK  [ ........ ]
                                   10 75 08 95 06 15 00 26        [ .u.....& ]
                                   FF 00 09 01 81 00 09 01        [ ........ ]
                                   91 00 C0 06 00 FF 09 02        [ ........ ]
SOF=747   I N  ADR=2   ENP=0 DATA0 A1 01 85 11 75 08 95 13   ACK  [ ....u... ]
                                   15 00 26 FF 00 09 02 81        [ ..&..... ]
                                   00 09 02 91 00 C0 06 00        [ ........ ]
                                   FF 09 04 A1 01 85 20 75        [ ...... u ]
          I N  ADR=2   ENP=0 DATA1 08 95 0E 15 00 26 FF 00   ACK  [ .....&.. ]
                                   09 41 81 00 09 41 91 00        [ .A...A.. ]
                                   85 21 95 1F 15 00 26 FF        [ .!....&. ]
                                   00 09 42 81 00 09 42 91        [ ..B...B. ]
          I N  ADR=2   ENP=0 DATA0 00 C0                     ACK  [ .. ]
          OUT  ADR=2   ENP=0 DATA1                           ACK
```

hathach, 2021-06-17:
@v9938 please put your log into its own txt and attached it instead of posting it here. Long log in the comment make it difficult follow the issue. Also please use above lsusb and hidrd to post your device configuration & hid descriptor. The log you post is not easy to read at all. If you are using windows try usbtreeview https://www.uwe-sieber.de/usbtreeview_e.html then post your device configuration and hid descriptor as attached txt.

dshadoff, 2021-06-17:
@hathach I tried with both of the mice, and the more-hid-host branch had problems; I have captured the logs.

### Mouse 1: Maxxster (wired mouse with side buttons):
When connected, it did not appear to respond at all.
I am attaching the device descriptions you mentioned previously (vid, usbhid), a capture of output by normal build (NOLOG), and a capture with output from build with LOG=3 (LOG3):
[maxxster_vid.txt](https://github.com/raspberrypi/pico-sdk/files/6673165/maxxster_vid.txt)
[maxxster_usbhid_18f8.txt](https://github.com/raspberrypi/pico-sdk/files/6673166/maxxster_usbhid_18f8.txt)
[maxxster_out_NOLOG.txt](https://github.com/raspberrypi/pico-sdk/files/6673168/maxxster_out_NOLOG.txt)
[maxxster_out_LOG3.txt](https://github.com/raspberrypi/pico-sdk/files/6673170/maxxster_out_LOG3.txt)

### Mouse 2: Logitech (unifying receiver):
Connected, but "couldn't find report info" for any reports when the mouse moved.
I am attaching the device descriptions you mentioned previously (vid, usbhid), a capture of output by normal build (NOLOG), and a capture with output from build with LOG=3 (LOG3):
[logitech_vid.txt](https://github.com/raspberrypi/pico-sdk/files/6673171/logitech_vid.txt)
[logitech_usbhid_046d.txt](https://github.com/raspberrypi/pico-sdk/files/6673172/logitech_usbhid_046d.txt)
[logitech_out_NOLOG.txt](https://github.com/raspberrypi/pico-sdk/files/6673174/logitech_out_NOLOG.txt)
[logitech_out_LOG3.txt](https://github.com/raspberrypi/pico-sdk/files/6673176/logitech_out_LOG3.txt)

v9938, 2021-06-17:
@hathach Thanks for the quick response.
I have attached logs from usbtreeview.

**Firmware version 012.010.00032(Model No:C-U007)**
[Unifiying_012_010_00032.txt](https://github.com/raspberrypi/pico-sdk/files/6673652/Unifiying_012_010_00032.txt)

**Firmware version 024.007.00030(Model No:C-U008)**
[Unifiying_024_007_00030.txt](https://github.com/raspberrypi/pico-sdk/files/6673653/Unifiying_024_007_00030.txt)

hathach, 2021-06-18:
@v9938 thanks for the configuration descriptor, I would also need the hid descriptor as well. I don't know how to that on windows. Maybe you should try to run usbtreeview with administrator mode to see if it could. Since we have the txt file now, could you remove your super long log in above post to make this issue easier to follow.

@dshadoff thank for the detail feedback, I will check this out later on.

v9938, 2021-06-19:
@hathach It seems to be difficult to dump the raw data of HID  on windows.
I have dumped the HID using usbhid-view.
If there is any other missing information, please reply.

**HID descriptor  version 012.010.00032(Model No:C-U007)**
[HIDdescriptor_for_C-U007.txt](https://github.com/raspberrypi/pico-sdk/files/6679835/HIDdescriptor_for_C-U007.txt)

**HID descriptor  version 024.007.00030(Model No:C-U008)**
[HIDdescriptor_for_C-U008.txt](https://github.com/raspberrypi/pico-sdk/files/6679838/HIDdescriptor_for_C-U008.txt)

dshadoff, 2021-06-20:
I looked again at this on the weekend (on the more-hid-host branch), also with a couple of other mice.

When compiled with LOG=1, the reports from Logitech (and also another wireless mouse) seemed to be providing correct data in the reports, and consistent with boot mouse format (i.e. buttons, x, y, scroll wheel - each 8-bit signed values).

The only problem in those cases was:
"Couldn't find the report info for this report !"
So if this issue can be corrected, it should work with most mice I think.

(However, the Maxxster appeared not to produce any reports... but I don't care as much about this mouse if the others are working)

hathach, 2021-06-21:
thanks @v9938 and @dshadoff for providing more info and log, I am currently switched to other paid work. Will check this out again later on whenever possible.

hathach, 2021-06-28:
@dshadoff @v9938 I have fixed an bug with host hid set protocol, and also update example to better use boot protocol. This should work with all keyboard/mouse with boot interface. Let me know if this would work for you.

dshadoff, 2021-06-28:
I tested with several mice - including the two mentioned earlier in this thread - and they all seemed to work, and to provide consistent results.  Thank You !
I will be happy to close this issue when the code gets integrated into pico-sdk (until then, perhaps it can stay open in order to inform other users about status).

The only thing I noticed, was surrounding the disconnection and reconnection of mouse devices - but that appears to be the same as issue #483 opened by @v9938 , so it can be tracked there.

I will continue to test with a variety of other devices I have; if I find any issues, I will open separate issues for tracking.

hathach, 2021-06-29:
thank for the feedback, look like I nailed it this time. For the release, it may take a bit of time, since I still want to do more tweak to host API but have been kept busy with other works.

hathach, 2021-06-30:
@v9938 let me know if the new code fox issue with your devices

v9938, 2021-06-30:
@hathach Sorry for the late feedback.
Unfortunately, there was no change in the behavior.
If there is anything I can do to help, please reply.

**Model No:C-U007**
```
HID device address = 1, instance = 0 is mounted
HID Interface Protocol = Keyboard
usbh_edpt_xfer 386: ASSERT FAILED
config_get_report_desc_complete 395: ASSERT FAILED
```

**Model No:C-U008**
```
*** PANIC ***

ep 0 in was already available
```

[HIDdescriptor_for_C-U008.txt](https://github.com/raspberrypi/pico-sdk/files/6679838/HIDdescriptor_for_C-U008.txt)

hathach, 2021-06-30:
@v9938 thanks for your feedback, seem like there is still issue :( . Could you also attach both log with `LOG=1` and `LOG=3` for each of your device.

v9938, 2021-07-02:
@hathach The LOG files are attached below.

**Model No:C-U007**
`LOG=1` [U007_LV1.log](https://github.com/raspberrypi/pico-sdk/files/6756550/U007_LV1.log) / `LOG=3` [U007_LV3.log](https://github.com/raspberrypi/pico-sdk/files/6756553/U007_LV3.log) , [U007_LV3_case2.log](https://github.com/raspberrypi/pico-sdk/files/6756565/U007_LV3_case2.log)

**Model No:C-U008**
`LOG=1` [U008_LV1.log](https://github.com/raspberrypi/pico-sdk/files/6756560/U008_LV1.log) / `LOG=3` [U008_LV3.log](https://github.com/raspberrypi/pico-sdk/files/6756563/U008_LV3.log)

dshadoff, 2021-07-02:
These logs are related to the disconnect/reconnect issue, issue #483 ... shouldn't they be tracked there instead ?
I interpreted @hathach 's comments to be "do you see issues with the data - OTHER THAN those tracked in issue 483 ?"

hathach, 2021-07-04:
indeed @v9938 this seems to be issue with connection, which can be either hardware and/or connection issue. To further troubleshoot it, I would suggest you to post picture of your hw set-up, also have shortest otg cable as possible etc... I don't see any abnormal connection issue with mine. I think people could then help you more with troubleshoot if it is hardware issue. I think my work here is done. Thanks both of you for testing out and giving feedback.

v9938, 2021-07-13:
@hathach  Sorry for the late response. There were several problems and it took a long time to prepare.
I have considered the possibility of my board being broken. I tried the same experiment again with a new pico board.
The result was no change :-(

The next picture shows the OTG cable. In both cables, the result was the same result.
[image]

The experimental setup was as shown in the next photo.
[image]

Maybe there is nothing wrong with the environment.

hathach, 2021-07-14:
I am not good at hardware, here is my set up that work relatively well. I use another pico as picoprobe (above) also providing 5V to target (below) via pin 40. 
[image]

dshadoff, 2021-11-03:
SDK version 1.3.0 release and the corresponding TinyUSB version are working fine for this.
I will close the issue now.

[1 more reply]

closed 2021-11-03 by dshadoff