raspberrypi/pico-sdk #446: elf2uf2.exe crashes [ninja build]

jonathangjertsen, 2021-05-25:
After doing nothing in particular, elf2uf2.exe has started crashing when I build it with ninja. Running it causes an error dialog to pop up saying

```
The procedure entry point _ZSt28__throw_bad_array_new_lengthv could not be located in the dynamic link library [...]\build\elf2uf2\elf2uf2.exe
```

Here is the full log

```
(base) c:\>git clone https://github.com/raspberrypi/pico-sdk.git
Cloning into 'pico-sdk'...
remote: Enumerating objects: 3446, done.
remote: Counting objects: 100% (976/976), done.
remote: Compressing objects: 100% (505/505), done.
remote: Total 3446 (delta 443), reused 794 (delta 347), pack-reused 2470
Receiving objects: 100% (3446/3446), 1.84 MiB | 2.40 MiB/s, done.
Resolving deltas: 100% (1550/1550), done.

(base) c:\>cd pico-sdk

(base) c:\pico-sdk>git submodule init
Submodule 'tinyusb' (https://github.com/raspberrypi/tinyusb.git) registered for path 'lib/tinyusb'

(base) c:\pico-sdk>git submodule update
Cloning into 'C:/pico-sdk/lib/tinyusb'...
Submodule path 'lib/tinyusb': checked out '11c23f88bf42f64ce14b8a7b0b2a4e207dc4dd12'

(base) c:\pico-sdk>mkdir build

(base) c:\pico-sdk>cd build

(base) c:\pico-sdk\build>cmake .. -G Ninja

(base) c:\pico-sdk\build>cmake .. -G Ninja
PICO_SDK_PATH is C:/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe
Defaulting PICO target board to pico since not specified.
Using board configuration from C:/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: C:/Users/dt/Miniconda3/python.exe (found version "3.7.2") found components:  Interpreter
TinyUSB available at C:/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
ELF2UF2 will need to be built
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/pico-sdk/build

(base) c:\pico-sdk\build>ninja kitchen_sink
[9/71] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Check for working C compiler: C:/mingw64/bin/gcc.exe
-- Check for working C compiler: C:/mingw64/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/mingw64/bin/c++.exe
-- Check for working CXX compiler: C:/mingw64/bin/c++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: C:/pico-sdk/build/elf2uf2
[10/71] Performing build step for 'ELF2UF2Build'
[1/2] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.obj
[2/2] Linking CXX executable elf2uf2.exe
[71/71] Linking CXX executable test\kitchen_sink\kitchen_sink.elf
FAILED: test/kitchen_sink/kitchen_sink.elf
cmd.exe /C "cd . && C:\PROGRA~2\GNUARM~1\102020~1\bin\AR10B2~1.EXE  -mcpu=cortex-m0plus -mthumb -O3 -DNDEBUG -Wl,--build-id=none -Wl,--wrap=__clzsi2 -Wl,--wrap=__clzdi2 -Wl,--wrap=__ctzsi2 -Wl,--wrap=__ctzdi2 -Wl,--wrap=__popcountsi2 -Wl,--wrap=__popcountdi2 -Wl,--wrap=__clz -Wl,--wrap=__clzl -Wl,--wrap=__clzll -Wl,--wrap=__aeabi_idiv -Wl,--wrap=__aeabi_idivmod -Wl,--wrap=__aeabi_ldivmod -Wl,--wrap=__aeabi_uidiv -Wl,--wrap=__aeabi_uidivmod -Wl,--wrap=__aeabi_uldivmod -Wl,--wrap=__aeabi_dadd -Wl,--wrap=__aeabi_ddiv -Wl,--wrap=__aeabi_dmul -Wl,--wrap=__aeabi_drsub -Wl,--wrap=__aeabi_dsub -Wl,--wrap=__aeabi_cdcmpeq -Wl,--wrap=__aeabi_cdrcmple -Wl,--wrap=__aeabi_cdcmple -Wl,--wrap=__aeabi_dcmpeq -Wl,--wrap=__aeabi_dcmplt -Wl,--wrap=__aeabi_dcmple -Wl,--wrap=__aeabi_dcmpge -Wl,--wrap=__aeabi_dcmpgt -Wl,--wrap=__aeabi_dcmpun -Wl,--wrap=__aeabi_i2d -Wl,--wrap=__aeabi_l2d -Wl,--wrap=__aeabi_ui2d -Wl,--wrap=__aeabi_ul2d -Wl,--wrap=__aeabi_d2iz -Wl,--wrap=__aeabi_d2lz -Wl,--wrap=__aeabi_d2uiz -Wl,--wrap=__aeabi_d2ulz -Wl,--wrap=__aeabi_d2f -Wl,--wrap=sqrt -Wl,--wrap=cos -Wl,--wrap=sin -Wl,--wrap=tan -Wl,--wrap=atan2 -Wl,--wrap=exp -Wl,--wrap=log -Wl,--wrap=ldexp -Wl,--wrap=copysign -Wl,--wrap=trunc -Wl,--wrap=floor -Wl,--wrap=ceil -Wl,--wrap=round -Wl,--wrap=sincos -Wl,--wrap=asin -Wl,--wrap=acos -Wl,--wrap=atan -Wl,--wrap=sinh -Wl,--wrap=cosh -Wl,--wrap=tanh -Wl,--wrap=asinh -Wl,--wrap=acosh -Wl,--wrap=atanh -Wl,--wrap=exp2 -Wl,--wrap=log2 -Wl,--wrap=exp10 -Wl,--wrap=log10 -Wl,--wrap=pow -Wl,--wrap=powint -Wl,--wrap=hypot -Wl,--wrap=cbrt -Wl,--wrap=fmod -Wl,--wrap=drem -Wl,--wrap=remainder -Wl,--wrap=remquo -Wl,--wrap=expm1 -Wl,--wrap=log1p -Wl,--wrap=fma -Wl,--wrap=__aeabi_fadd -Wl,--wrap=__aeabi_fdiv -Wl,--wrap=__aeabi_fmul -Wl,--wrap=__aeabi_frsub -Wl,--wrap=__aeabi_fsub -Wl,--wrap=__aeabi_cfcmpeq -Wl,--wrap=__aeabi_cfrcmple -Wl,--wrap=__aeabi_cfcmple -Wl,--wrap=__aeabi_fcmpeq -Wl,--wrap=__aeabi_fcmplt -Wl,--wrap=__aeabi_fcmple -Wl,--wrap=__aeabi_fcmpge -Wl,--wrap=__aeabi_fcmpgt -Wl,--wrap=__aeabi_fcmpun -Wl,--wrap=__aeabi_i2f -Wl,--wrap=__aeabi_l2f -Wl,--wrap=__aeabi_ui2f -Wl,--wrap=__aeabi_ul2f -Wl,--wrap=__aeabi_f2iz -Wl,--wrap=__aeabi_f2lz -Wl,--wrap=__aeabi_f2uiz -Wl,--wrap=__aeabi_f2ulz -Wl,--wrap=__aeabi_f2d -Wl,--wrap=sqrtf -Wl,--wrap=cosf -Wl,--wrap=sinf -Wl,--wrap=tanf -Wl,--wrap=atan2f -Wl,--wrap=expf -Wl,--wrap=logf -Wl,--wrap=ldexpf -Wl,--wrap=copysignf -Wl,--wrap=truncf -Wl,--wrap=floorf -Wl,--wrap=ceilf -Wl,--wrap=roundf -Wl,--wrap=sincosf -Wl,--wrap=asinf -Wl,--wrap=acosf -Wl,--wrap=atanf -Wl,--wrap=sinhf -Wl,--wrap=coshf -Wl,--wrap=tanhf -Wl,--wrap=asinhf -Wl,--wrap=acoshf -Wl,--wrap=atanhf -Wl,--wrap=exp2f -Wl,--wrap=log2f -Wl,--wrap=exp10f -Wl,--wrap=log10f -Wl,--wrap=powf -Wl,--wrap=powintf -Wl,--wrap=hypotf -Wl,--wrap=cbrtf -Wl,--wrap=fmodf -Wl,--wrap=dremf -Wl,--wrap=remainderf -Wl,--wrap=remquof -Wl,--wrap=expm1f -Wl,--wrap=log1pf -Wl,--wrap=fmaf -Wl,--wrap=__aeabi_lmul -Wl,--wrap=malloc -Wl,--wrap=calloc -Wl,--wrap=free -Wl,--wrap=memcpy -Wl,--wrap=memset -Wl,--wrap=__aeabi_memcpy -Wl,--wrap=__aeabi_memset -Wl,--wrap=__aeabi_memcpy4 -Wl,--wrap=__aeabi_memset4 -Wl,--wrap=__aeabi_memcpy8 -Wl,--wrap=__aeabi_memset8 --specs=nosys.specs -Wl,--wrap=sprintf -Wl,--wrap=snprintf -Wl,--wrap=vsnprintf -Wl,-Map=kitchen_sink.elf.map -Wl,--script=C:/pico-sdk/src/rp2_common/pico_standard_link/memmap_default.ld -Wl,--gc-sections -Wl,--wrap=printf -Wl,--wrap=vprintf -Wl,--wrap=puts -Wl,--wrap=putchar @CMakeFiles\kitchen_sink.rsp  -o test\kitchen_sink\kitchen_sink.elf  && cmd.exe /C "cd /D C:\pico-sdk\build\test\kitchen_sink && "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-objcopy.exe" -Oihex kitchen_sink.elf kitchen_sink.hex && cd /D C:\pico-sdk\build\test\kitchen_sink && "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-objcopy.exe" -Obinary kitchen_sink.elf kitchen_sink.bin && cd /D C:\pico-sdk\build\test\kitchen_sink && "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-objdump.exe" -h kitchen_sink.elf >kitchen_sink.dis && "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-objdump.exe" -d kitchen_sink.elf >>kitchen_sink.dis && cd /D C:\pico-sdk\build\test\kitchen_sink && C:\pico-sdk\build\elf2uf2\elf2uf2 kitchen_sink.elf kitchen_sink.uf2""
ninja: build stopped: subcommand failed.
```

There is no such error if I build with nmake.

I know ninja is not really supported but does anyone have a clue what might have happened here?

jonathangjertsen, 2021-05-25:
Possible clue here https://github.com/stan-dev/pystan/issues/294

In that case it was solved by not using conda, however that did not change anything in my case.

Seems to be a problem with wrong versions of libstdc++, with `throw_bad_array_new_lengthv` having been introduced quite recently

jonathangjertsen, 2021-05-25:
Workaround:

* Make a different build folder using nmake and build elf2uf2.exe there
* Move elf2uf2.exe to some folder that's in the PATH
* In the original build folder, run `cmake .. -DELF2UF2_FOUND=1`
* Run ninja, it will now use the external elf2uf2.exe binary

kilograham, 2021-05-25:
Not abundantly clear why ninja/cmake would affect it (esp if it is a gcc 11 issue) is ninja build using the same gcc version, or somehow picking up a different one. might be worth comparing verbose build output. 

I built with ninja on windows only yesterday, as I had merged a bunch of things into `develop`, and that worked, but I'm certainly not using gcc11.

Re-open if you have something more

closed 2021-05-25 by kilograham

jonathangjertsen, 2022-01-14:
I encountered this again while setting up a new pico project, but since the above workaround works I'm not going look further into it. One thing though: when elf2uf2 exists in the PATH, shouldn't it just work without setting `-DELF2UF2_FOUND=1`?

jonathangjertsen, 2022-07-17:
I encountered this again when starting a new project and dug into it a bit more. It is an instance of this issue: https://stackoverflow.com/a/38708587

I can circumvent the issue by adding this to the CMakeLists.txt in the elf2uf2 folder

```cmake
target_link_options(elf2uf2 PRIVATE -static-libstdc++)
```

Similarly for pioasm

```cmake
target_link_options(pioasm PRIVATE -static-libstdc++)
```

I couldn't find a workaround that I could do locally in my project to achieve the same thing