raspberrypi/pico-sdk #474: Compile error when installing with pico_setup.sh

fivdi, 2021-06-03:
Trying to install with pico_setup.sh (https://raw.githubusercontent.com/raspberrypi/pico-setup/master/pico_setup.sh) results in this error:

```
Scanning dependencies of target picoprobe_probe_pio_h
[ 22%] Generating probe.pio.h
[ 22%] Built target picoprobe_probe_pio_h
Scanning dependencies of target picoprobe
[ 23%] Building C object CMakeFiles/picoprobe.dir/src/led.c.obj
[ 25%] Building C object CMakeFiles/picoprobe.dir/src/main.c.obj
[ 25%] Building C object CMakeFiles/picoprobe.dir/src/usb_descriptors.c.obj
[ 26%] Building C object CMakeFiles/picoprobe.dir/src/probe.c.obj
[ 27%] Building C object CMakeFiles/picoprobe.dir/src/cdc_uart.c.obj
[ 28%] Building C object CMakeFiles/picoprobe.dir/src/get_serial.c.obj
[ 29%] Building C object CMakeFiles/picoprobe.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
In file included from /home/pi/pico/picoprobe/src/get_serial.c:27:0:
/home/pi/pico/pico-sdk/src/rp2_common/pico_unique_id/include/pico/unique_id.h:72:52: error: unknown type name 'uint'; did you mean 'int'?
 void pico_get_unique_board_id_string(char *id_out, uint len);
                                                    ^~~~
                                                    int
make[2]: *** [CMakeFiles/picoprobe.dir/build.make:128: CMakeFiles/picoprobe.dir/src/get_serial.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1515: CMakeFiles/picoprobe.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

I think this is happening because the declaratrion of `pico_get_unique_board_id_string` in `header unique_id.h` ([see here](https://github.com/raspberrypi/pico-sdk/blob/1.2.0/src/rp2_common/pico_unique_id/include/pico/unique_id.h#L72)) uses type `uint` but the header `unique_id.h` doesn't include `pico.h` or `types.h` and therfore doesn't know what a `uint` is.

kilograham, 2021-06-03:
merged here, but i have added a pico.h include to picoprobe for the interim

kilograham, 2021-06-03:
merged into develop

PR #475 (2021-06-03): include pico.h so that uint is defined
Fixes #474
closed 2021-06-03 by kilograham