raspberrypi/pico-sdk #483: Logitech USB receiver connection failure

v9938, 2021-06-12:
Split from #422.

Logitech USB receiver connection failed for the second time.
When using a Logitech USB receiver with tinyUSB cdc_msc_hid example.
The first connection was okay, but the second connection was PANIC.

Board:     Raspberry Pi pico
SDK:        version 1.2.0
tinyUSB:  host-rp2040-double-buffer

The serial log is as follows
CFG_TUSB_DEBUG=1
[LV1_logitech_pico_log_20210612.txt](https://github.com/raspberrypi/pico-sdk/files/6642775/LV1_logitech_pico_log_20210612.txt)

CFG_TUSB_DEBUG=3
[LV3_logitech_pico_log_20210612.txt](https://github.com/raspberrypi/pico-sdk/files/6642374/logitech_pico_log_20210612.txt)

The test scenario followed the steps below.
1.Connect Logitech receiver.
2.Power ON 
3.Type 123456790 on the keyboard
4.Press and release the left mouse button.
5.Press and release the right mouse button.
6.Disconnect Logitech receiver
7.Connect Logitech receiver

If you need any other information about this issue, please request it.

dshadoff, 2021-08-12:
As I have been following the developments on this issue and in general on TinyUSB and pico-sdk, it seems to me that this issue is all about disconnection/reconnection of the receiver (I have seen a similar behaviour, but am not in a position to test).

Also, based on its description, it seems to me that pull request "Add Rp2040 suspend & resume support #1020" on the TinyUSB repository is likely to address this scenario (once it is accepted).

@hathach is this a fair assessment ?

hathach, 2021-08-13:
I cannot tell, I never reproduce this issue, it probably purely physical/hw issue. If you could reproduce it, please try with the PR to see if that helps.

dshadoff, 2021-11-03:
Based on the version of TinyUSB included with the pico-sdk version 1.3.0, I no longer see an issue when RP2040 acts as USB host and HID device is disconnected and reconnected.

Original poster should retry with the new SDK.
(Note that several other changes have occurred in the host HID example code; you should update your code to match).

kilograham, 2021-11-03:
cc @liamfraser

liamfraser, 2021-11-05:
waiting on @v9938 to retry as I don't have this hardware available

v9938, 2021-11-05:
I'll check it out this weekend. Please give me some time.

v9938, 2021-11-10:
I have confirmed this with SDK1.30.
This is unfortunate infomation.
The panic still occurs.

Using a packet analyzer, I found that the bus transaction went into an invalid state at a certain point.

[image][image]

hathach, 2021-11-11:
@v9938 this is started as physical issue since your mouse `connected --> disconnected --> connected`. However, as a good stack, TinyUSB should have handled this scenario. So yeah, it is a bug, but since I don't have your device to reproduce and host stack still needs lots more work. This will probably be opened for a while.

v9938, 2021-11-11:
@hathach There is no problem with that response.
I understand you have a lot of work to do.

I would like to give the Logitech C-U008 receiver as a gift.
So,I am trying to get a spare unit.

I'll send a direct message later.

hathach, 2021-11-11:
@v9938 thank you for your willingness, though it is not really necessary, issue like this one will likely be fixed when host stack is improved over time. It is device-specific and not an major issue at all.

hathach, 2021-11-30:
you may want to try out https://github.com/hathach/tinyusb/pull/1193 , it probably fixes your issue.

kilograham, 2022-07-14:
should be fixed in next SDK release which will pick up TinyUSB 0.14.0

kilograham, 2023-02-08:
TinyUSB updated in develop branch.

closed 2023-02-08 by kilograham