raspberrypi/pico-sdk #487: Possibility to invert UART RX and TX ?

nordseele, 2021-06-17:
Hi, 
Is it possible to invert the hardware RX and TX of the Pico ? (UART1, gpio 8 and 9) ?

I made a stupid mistake while designing my PCB and now I can't connect an ESP-8266 to the Pico without rewiring Tx to Rx and Rx to TX. 

Before I send a new revision of my board to the factory, I'd like to know if there's a way to do this within the Pico SDK.
I've found [a thread](https://www.raspberrypi.org/forums/viewtopic.php?t=306053) mentioning a possibility to do this with [MicroPython](http://docs.micropython.org/en/v1.15/library/machine.UART.html?highlight=uart%20invert) but I can't find a reference to "invert" in the SDK documentation. 

Thanks!

lurch, 2021-06-17:
If you're using one of the two the "hardware UARTs" (ARM PL011) then the pins for these can't be remapped, but if you're using "software UARTs" (running on the PIO), then PIO has the flexibility to use any pins you like. (take a look at https://github.com/raspberrypi/pico-examples/tree/master/pio/uart_tx and https://github.com/raspberrypi/pico-examples/tree/master/pio/uart_rx )

nordseele, 2021-06-17:
Thanks a lot for your reply. 

Yes, unfortunately, I'm using the hardware UART for the ESP on the PCB. Too bad, I guess I'll just make a new revision... but anyway it's very nice learning about the PIO system, I might use that someday. 

Thanks again :-)

closed 2021-06-17 by nordseele