raspberrypi/pico-sdk #488: pwm_set_enabled(slice, false) can leave the PWM pin high

DavePutz, 2021-06-24:
Apparently if calling pwm_set_enabled() to disable PWM is done too soon after a pulse is ended it can leave the pin high.  Inserting a short delay before the call seems to resolve the issue.
Code:
```
#include "pico/stdlib.h"
#include "hardware/pwm.h"
#include "pico/time.h"
#include <stdio.h>

#define NUM_PULSES 40
#define PIN_NUM 28

static volatile uint16_t pulse_index = 0;
static uint16_t pulse_length;
static uint32_t slice_num;

int64_t pulseout_interrupt_handler(alarm_id_t id, void *user_data) {
       pulse_index++;
       // Turn pwm pin off
       pwm_set_chan_level(slice_num,PWM_CHAN_A,0);
       if (pulse_index >= pulse_length) {
	    return 0;
       }
       add_alarm_in_us(NUM_PULSES, pulseout_interrupt_handler, NULL, false);
       if (pulse_index % 2 == 0) {
	    pwm_set_chan_level(slice_num,PWM_CHAN_A,1);
	}
	return 0;
}	

int main() {

    stdio_init_all();
    sleep_ms(5000);
    printf("Starting PWM\n");
    // Pin GPIO PIN_NUM allocated to PWM
    gpio_set_function(PIN_NUM, GPIO_FUNC_PWM);
    slice_num = pwm_gpio_to_slice_num(PIN_NUM);
    pulse_length = 50;

    pwm_set_wrap(slice_num, 3);
    // Set channel A output high for one cycle before dropping
    pwm_set_chan_level(slice_num, PWM_CHAN_A, 1);
    // Start running
    while (true) {
	add_alarm_in_us(NUM_PULSES,pulseout_interrupt_handler, NULL, false);
	pwm_set_chan_level(slice_num, PWM_CHAN_A,1);
        pwm_set_enabled(slice_num, true);
        while (pulse_index < NUM_PULSES ) {
	}
//        sleep_us(50);
        pwm_set_enabled(slice_num, false);
        sleep_ms(1000);
	pulse_index = 0;
    }
}
```
The frequency of leaving the pin high varies - it looks like about 1/4 of the time overall. Uncommenting the short sleep looks like it resolves the pin level issue.

Wren6991, 2021-07-06:
> pwm_set_enabled(slice, false) can leave the PWM pin high

Yes that's right, it stops the PWM counter at its current count. Depending on exactly when this is, the pin may be high or low. If you want to force the pin low, your could:

- Set PWM level to 0, which gives you a glitch-free always-low state
- Use GPIO overrides to force the pin low independent of what the PWM is doing

Wren6991, 2021-07-06:
> The frequency of leaving the pin high varies - it looks like about 1/4 of the time overall.

You're toggling the pin with 1/4 duty cycle at around 30 MHz, so stopping at a random time would give you around a 1/4 chance of stopping when the pin is high. To confirm this is happening, you could try printing the value of `pwm_get_counter(slice_num)` after halting, and you should see the pin high if and only if the count is `1`.

> Uncommenting the short sleep looks like it resolves the pin level issue

If I had to guess what is going on with your sleep call, I'd say the sleep is adding some jitter to the exact cycle when the PWM is stopped, and taking `jitter % 4` you get an apparently random pin level when halting. On the other hand, without the sleep call, you always halt with exactly the same timing, and this happens to be on a cycle where the pin is low.

lurch, 2021-07-06:
Might be worth adding an extra comment to https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_pwm/include/hardware/pwm.h#L445 ?

Wren6991, 2021-07-19:
Hopefully those last answers cleared things up. I've added a PR for some comments in pwm.h.

Note one way to force the pins low when a slice is halted is:

```
#include "hardware/structs/pwm.h"

void force_pwm_pins_low(uint slice_num) {
    pwm_hw->slice[slice_num].ctr = 0;
    pwm_hw->slice[slice_num].cc = 0;
    pwm_hw->slice[slice_num].top = 0;
    pwm_hw->slice[slice_num].div = 1u << PWM_CH0_DIV_INT_LSB;
    pwm_hw->slice[slice_num].csr = PWM_CH0_CSR_EN_BITS;
    pwm_hw->slice[slice_num].csr = 0;
}
```

This momentarily runs the PWM with a 1-cycle period and zero output levels.

closed 2021-07-19 by Wren6991