raspberrypi/pico-sdk #489: 64bit division is unsafe against float division in interrupts

elehobica, 2021-06-26:
Under pico-sdk 1.2.0,
64bit division in main function could result incorrect when float division is performed in interrupt asynchronously.

test code is as below (same on https://github.com/elehobica/pico_divider_test/blob/f56d42cfb7dd13848f1726b24ba8fb9ca411cd37/pico_divider_test.cpp)

```
#include <cstdio>
#include <cstdlib>
#include "pico/stdlib.h"
#include "hardware/divider.h"

static repeating_timer_t timer;
static const uint32_t SampRateHz = 44100;
static const float SampRateHz_div_1000 = 44.1;
uint32_t dummy = 0;

static inline uint32_t _millis(void)
{
    return to_ms_since_boot(get_absolute_time());
}

bool timer_callback(repeating_timer_t *rt) {
    uint32_t millis;
    uint32_t sec;
    for (int i = 0; i < 100; i++) {
        uint32_t millis_ref = (rand() & 0x007fffff) + 1;
        uint32_t samplesPlayed = (uint32_t) ((float) millis_ref * SampRateHz_div_1000);

        //dummy += (uint32_t) ((uint64_t) samplesPlayed * 1000 / SampRateHz);
        dummy += (uint32_t) ((float) samplesPlayed * 1000.0f / (float) SampRateHz); // seems only float division affects to 64bit division in main badly
        //dummy += (samplesPlayed ) / SampRateHz;
    }
    return true; // keep repeating
}

int main()
{
    srand(0);
    stdio_init_all();
    sleep_ms(1000);

    puts("START");
    if (!add_repeating_timer_us(-1000, timer_callback, nullptr, &timer)) {
        return 0;
    }

    int count = 0;
    while (true) {
        //uint32_t millis_ref = _millis(); // failure occurs only when dividend is large enough
        uint32_t millis_ref = (rand() & 0x007fffff) + 1;
        uint32_t samplesPlayed = (uint32_t) ((float) millis_ref * SampRateHz_div_1000);
        uint32_t millis = (uint32_t) ((uint64_t) samplesPlayed * 1000 / SampRateHz);

        if (millis < millis_ref - 1 || millis > millis_ref + 1) {
            printf("ERROR [calc_64bit][%d] samplesPlayed = %lu, millis_ref = %d, millis = %lu\n", count, samplesPlayed, millis_ref, millis);
        }
        count++;
        if (_millis() > 10 * 1000) { break; }
    }

    printf("dummy = %d\n", dummy); // Not to delete dummy by optimization
    printf("END\n");
    return 0;
}
```

I have checked only the case of dividend: uint64_t and divisor: uint32_t.
As far as I confirmed, it seems only float division in interrupts affects 64bit division in main and failure happens only when 64bit dividend is large enough.

Thank you in advance for your review.

kilograham, 2021-06-27:
thanks for this, that was a particularly insidious little bug (actually pre-existing b4 the float stuff)

elehobica, 2021-06-28:
Thanks for quick fix. Confirmed that this issue is cleared.

closed 2021-06-28 by elehobica