raspberrypi/pico-sdk #501: Can't build the sdk [error: 'ix' is used uninitialized]

Frandcam, 2021-06-30:
Hello, 

I can't compile the latest SDK. With my Linux version 

`Linux dan-acer 5.10.42-1-MANJARO #1 SMP PREEMPT Thu Jun 3 14:37:11 UTC 2021 x86_64 GNU/Linux
`
And the compiler 

`/usr/bin/arm-none-eabi-gcc version 11.1.0 (Arch Repository)`

There is strange error 

```
/home/daniil/pico/pico-sdk/src/rp2_common/pico_double/double_math.c: In function 'dcopysign':
/home/daniil/pico/pico-sdk/src/rp2_common/pico_double/double_math.c:92:12: error: 'ix' is used uninitialized [-Werror=uninitialized]
   92 |     return *(double*)&ix;
      |            ^~~~~~~~~~~~~
/home/daniil/pico/pico-sdk/src/rp2_common/pico_double/double_math.c:90:10: note: 'ix' declared here
   90 |     ui64 ix=*(ui64*)&x,iy=*(ui64*)&y;
      |          ^~
/home/daniil/pico/pico-sdk/src/rp2_common/pico_double/double_math.c: In function 'dneg':
/home/daniil/pico/pico-sdk/src/rp2_common/pico_double/double_math.c:146:12: error: 'ix' is used uninitialized [-Werror=uninitialized]
  146 |     return *(double*)&ix;
      |            ^~~~~~~~~~~~~
/home/daniil/pico/pico-sdk/src/rp2_common/pico_double/double_math.c:144:10: note: 'ix' declared here
  144 |     ui64 ix=*(ui64*)&x;
      |          ^~
cc1: all warnings being treated as errors
make[2]: *** [test/kitchen_sink/CMakeFiles/kitchen_sink_no_flash.dir/build.make:617: test/kitchen_sink/CMakeFiles/kitchen_sink_no_flash.dir/__/__/src/rp2_common/pico_double/double_math.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1757: test/kitchen_sink/CMakeFiles/kitchen_sink_no_flash.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

kilograham, 2021-06-30:
that is odd; looks like a compiler bug? I presume this has nothing to do with what version of the SDK you try to compile.

I will open an issue to check gcc 11 with the next SDK version, but presumably you can compile with 10.x for now.

Frandcam, 2021-06-30:
Ok. Thank You

kilograham, 2021-06-30:
actually i'll use this issue!

lurch, 2021-06-30:
GCC 11 isn't on https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads yet, so perhaps there are arm-none-eabi bugs in the ArchLinux build of arm-none-eabi-gcc version 11? :shrug:

dpgeorge, 2021-07-02:
GCC 11 does have a lot of new warnings (eg https://github.com/micropython/micropython/commit/211c3e41f12295f08a883ac326386ecb5bc8d68e) and with this particular one I can't see why the warning is there, it's strange.

kilograham, 2021-07-02:
Yeah, i mean it might be unhappy about the aliasing, but the particular error itself smacks of internal confusion

dpgeorge, 2021-07-02:
Use of a union could probably get around the warning.  Better still would be to fix GCC (if it really is broken).

PM1001, 2021-07-20:
You could also wrap it in

```
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wuninitialized"

/*Offending code goes here*/

#pragma GCC diagnostic pop
```
I don't know if it is really an error, but this at least let's you compile on gcc11.

ConnorRigby, 2021-09-06:
Any word on this? i'm getting the same issue

kilograham, 2021-09-06:
I believe it may be fixed with gcc 11.2

kilograham, 2021-10-06:
where r u guys getting arm-non-eabi-gcc 11 to test with?

dpgeorge, 2021-10-06:
> where r u guys getting arm-non-eabi-gcc 11 to test with?

I'm using Arch Linux with the standard arm-none-eabi-gcc package, which is currently:
```
community/arm-none-eabi-gcc 11.2.0-1 [installed]
    The GNU Compiler Collection - cross compiler for ARM EABI (bare-metal) target
```
When building for the rp2040 I get:
```
...
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
...
[ 85%] Building C object CMakeFiles/.../lib/pico-sdk/src/rp2_common/pico_double/double_math.c.obj
.../lib/pico-sdk/src/rp2_common/pico_double/double_math.c: In function 'dcopysign':
.../lib/pico-sdk/src/rp2_common/pico_double/double_math.c:92:12: warning: 'ix' is used uninitialized [-Wuninitialized]
   92 |     return *(double*)&ix;
      |            ^~~~~~~~~~~~~
.../lib/pico-sdk/src/rp2_common/pico_double/double_math.c:90:10: note: 'ix' declared here
   90 |     ui64 ix=*(ui64*)&x,iy=*(ui64*)&y;
      |          ^~
.../lib/pico-sdk/src/rp2_common/pico_double/double_math.c: In function 'dneg':
.../lib/pico-sdk/src/rp2_common/pico_double/double_math.c:146:12: warning: 'ix' is used uninitialized [-Wuninitialized]
  146 |     return *(double*)&ix;
      |            ^~~~~~~~~~~~~
.../lib/pico-sdk/src/rp2_common/pico_double/double_math.c:144:10: note: 'ix' declared here
  144 |     ui64 ix=*(ui64*)&x;
      |          ^~
...
```

kilograham, 2021-10-07:
ah, so not fixed in 11.2, i guess i can spin up a docker container

kilograham, 2021-10-07:
well this was a strict aliasing warning before, so union it is (which curiously produces better code actually)

kilograham, 2021-10-12:
merged into develop

closed 2021-10-12 by kilograham