raspberrypi/pico-sdk #505: Can't get i2c interrupt working on core1.

Aemulation, 2021-07-12:
Hello,

I have have a pico working as a i2c slave, using i2c interupts.
I have tried to let core1 do the i2c communication but it seems as if core1 doesn't receive the interrupts.

I have a simplified version of the code below.

```c++
#include <iostream>
#include <stdio.h>

#include "pico/stdlib.h"
#include "pico/multicore.h"
#include "hardware/i2c.h"
#include "hardware/irq.h"
#include "hardware/uart.h"

#define I2C0_SLAVE_ADDR 0x04

#define GPIO_SDA0 0
#define GPIO_SCK0 1

#define I2C_INTR_MASK_RX_FULL 0x00000004
#define I2C_INTR_MASK_READ_REQ 0x00000020

volatile uint32_t *const I2C0_DATA_CMD
    = reinterpret_cast<uint32_t *const>(I2C0_BASE + 0x10);
volatile uint32_t *const I2C0_INTR_STAT
    = reinterpret_cast<uint32_t *const>(I2C0_BASE + 0x2C);
volatile uint32_t *const I2C0_INTR_MASK
    = reinterpret_cast<uint32_t *const>(I2C0_BASE + 0x30);
volatile uint32_t *const I2C0_CLR_RD_REQ
    = reinterpret_cast<uint32_t *const>(I2C0_BASE + 0x50);

void i2c0_irq_handler() {
    std::cout << "test" << std::endl;
}

void core1_entry() {
    i2c_init(i2c0, 100 * 1000);
    i2c_set_slave_mode(i2c0, true, I2C0_SLAVE_ADDR);

    gpio_set_function(GPIO_SDA0, GPIO_FUNC_I2C);
    gpio_set_function(GPIO_SCK0, GPIO_FUNC_I2C);
    gpio_pull_up(GPIO_SDA0);
    gpio_pull_up(GPIO_SCK0);

    /* Enable the interrupts we want. */
    *I2C0_INTR_MASK = (I2C_INTR_MASK_READ_REQ | I2C_INTR_MASK_RX_FULL);

    /* Set up the interrupt handlers. */
    irq_set_exclusive_handler(I2C0_IRQ, i2c0_irq_handler);
    /* Enable I2C interrupts. */
    irq_set_enabled(I2C0_IRQ, true);

    while (true) {
        tight_loop_contents();
    }
}

int main() {
    stdio_init_all();

    //multicore_launch_core1(&core1_entry);
    core1_entry();

    while (true) {
        tight_loop_contents();
    }
}
```

When I let core0 run the function `core1_entry`, 'test' is printed a couple of times when the i2c master tries to communicate.
But when I use `multicore_launch_core1(&core1_entry);` instead of `core1_entry`, in the main function, nothing gets printed.

Am I using the multicore function wrong or is there some other reason `i2c0_irq_handler` doesn't get called?

kilograham, 2021-07-12:
it would be worth checking whether the problem is with the printing or the I2C IRQ - perhaps try toggling a LED to check?

Aemulation, 2021-07-12:
I tried turning on the LED on GPIO25 and I got the same result.
Works fine on core0 but not on core1.

lurch, 2021-07-12:
On an unrelated side-note, the constants that you use here:
```c++
volatile uint32_t *const I2C0_DATA_CMD
    = reinterpret_cast<uint32_t *const>(I2C0_BASE + 0x10);
volatile uint32_t *const I2C0_INTR_STAT
    = reinterpret_cast<uint32_t *const>(I2C0_BASE + 0x2C);
volatile uint32_t *const I2C0_INTR_MASK
    = reinterpret_cast<uint32_t *const>(I2C0_BASE + 0x30);
volatile uint32_t *const I2C0_CLR_RD_REQ
    = reinterpret_cast<uint32_t *const>(I2C0_BASE + 0x50);
```
can also be found in https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_regs/include/hardware/regs/i2c.h
```c
#define I2C_IC_DATA_CMD_OFFSET _u(0x00000010)
#define I2C_IC_INTR_STAT_OFFSET _u(0x0000002c)
#define I2C_IC_INTR_MASK_OFFSET _u(0x00000030)
#define I2C_IC_CLR_RD_REQ_OFFSET _u(0x00000050)
```

Alternatively, you may prefer to use `i2c0_hw` from https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_structs/include/hardware/structs/i2c.h

Aemulation, 2021-07-12:
Thanks, `i2c0_hw` looks a lot easier to use than what I have now.

kilograham, 2021-07-12:
Note also feel free to open an issue for any missing helper functions in i2c.h (seems like we are missing some IRQ setup related ones, that we generally have for other h/w)

Aemulation, 2021-07-12:
It's true that there aren't many helper functions for i2c slave operations, so I might do just that.

kilograham, 2021-07-12:
> I tried turning on the LED on GPIO25 and I got the same result.
> Works fine on core0 but not on core1.

just double checking on that one that you did the `gpio_init` for the LED pin from core1 when you did that?

Aemulation, 2021-07-12:
I did, I put both `gpio_init` and `gpio_set_dir` in the `core1_entry` function.

Aemulation, 2021-07-12:
It seems the problem only appears when I upload the .elf binary using the SWD port (using a raspberry pi 4).
Uploading the .uf2 binary using usb mass storage transfer gives no problems.

The problem is probably related to #386.
@kilograham mentioned there that you must do a `monitor reset init` everytime beforehand, and that does seem to work as well in my case.

Can that however be reduced to a single command? I have been using
`openocd -f "interface/raspberrypi-swd.cfg" -f "target/rp2040.cfg" -c "program multicore.elf verify reset exit"`
this whole time, and that alone does not seem to work.

closed 2021-07-23 by Aemulation