raspberrypi/pico-sdk #506: elf2uf2 raise "The input file has no memory pages" for a valid elf file

GorgonMeducer, 2021-07-13:
[reproducer.zip](https://github.com/raspberrypi/pico-sdk/files/6812321/reproducer.zip)

Hi, 

I manage to generate a valid elf file (see attachment), but the elf2uf2 raises the "The input file has no memory pages" issue.

The elf is generated from arm compiler 6.16 and the MDK project can be found here:
https://github.com/GorgonMeducer/Pico_Template

In general, the map file shows that the elf contains valid content, so you don't have to use the MDK project. 

By the way, I know that pico-SDK only support GCC, but with a simple wrapper, I make it work for Arm Compiler 6. I was about to try the image on pico, but the issue coming from elf2uf2 (built for win32) blocks me.

Please help. Thank you.

kilograham, 2021-07-14:
Ah, yes, it seems that elf2uf2 assumes that the ELF program header immediately follows the ELF file header, which it usually does (but doesn't have to)...

You can use SWD debug to download the ELF directly, or use https://github.com/microsoft/uf2/tree/master/utils to convert the BIN file to a UF2 as a workaround.

That said, your binary isn't going to work anyway, as it does not have a second stage boot from `0x10000000-0x10000100`

GorgonMeducer, 2021-07-14:
Hi Kilograham,

What a quick reply! 

I will try the bin2uf2 solution, thank you, and I hope the elf2uf2 issue could be fixed soon. 

About the second stage boot, which files should I bring into compilation? do you have any suggestions?

Thank you in advance.

lurch, 2021-07-14:
> About the second stage boot, which files should I bring into compilation? do you have any suggestions?

That depends which board you're using, see https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards

If you're using a Raspberry Pi Pico, see https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/pico.h#L67 and https://github.com/raspberrypi/pico-sdk/tree/master/src/rp2_common/boot_stage2

GorgonMeducer, 2021-07-14:
I am using Raspberry Pi Pico. 
I will try https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/pico.h#L67 and https://github.com/raspberrypi/pico-sdk/tree/master/src/rp2_common/boot_stage2 

Cheers.

lurch, 2021-07-14:
There's also in-depth info about the boot stage 2 in section 2.8 of https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf (which may be useful if you're rolling your own build system)

GorgonMeducer, 2021-07-14:
Yes, I am reading the sdk manual and rp2040 datasheet. I was trying to ignore those parts (to save time...), but it looks like I shouldn't.

GorgonMeducer, 2021-07-14:
By the way, the SDK manual is entirely focusing on CMake and provide limited information for people who want to use a different build system. For example, I thought Appendix D: Board Configuration might help, but it doesn't. It is sad.

If there is any chapter to help people understand the compilation relationship and how the header files are included (obviously there are plenty of decision makings are done by CMake), it will help a lot.

Don't you think people would like to try SDK on IAR, LLVM, arm compiler 5 and arm compiler 6? Not only to mention that the current version of arm GCC is way behind the LLVM, armcc and armclang in terms of performance and code size.

Cheers

GorgonMeducer, 2021-07-14:
I move the discussion about lack of documentation to a dedicated issue.
Let's focus on the uf2 conversion issue here.

aallan, 2021-07-14:
> I move the discussion about lack of documentation to a dedicated issue.

I transferred the issue to the feedback repo, raspberrypi/pico-feedback#176. Not an SDK issue.

GorgonMeducer, 2021-07-14:
sure

closed 2021-07-14 by GorgonMeducer

lurch, 2021-07-15:
This was fixed by #512 
(just adding this comment for cross-referencing purposes)

commit ccc0ba56 (2021-07-14): elf2uf2: correctly parse ELF files where the program header is not directly following the file header

fixes #506