raspberrypi/pico-sdk #510: error: invalid use of non-static data member ‘address_range::type’, and other errors when building blink.c

xayhewalo, 2021-07-14:
I'm following the C/C++ SDK [getting started guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) on Ubuntu 16.04 and CMake 3.20.5. I'm failing to build when running the steps in Section 3.1 (page 9).

The main problem seems to be `address_range::type` in `/pico-sdk/tools/elf2uf2/main.cpp`. Below are some errors pulled from the build log.

```
/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:73:64: error: invalid use of non-static data member ‘address_range::type’
     address_range(MAIN_RAM_START, MAIN_RAM_END, address_range::type::NO_CONTENTS
                                                                ^
```
```
/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:73:64: error: invalid use of non-static data member ‘address_range::type’
     address_range(MAIN_RAM_START, MAIN_RAM_END, address_range::type::NO_CONTENTS
                                                                ^
```
```
/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:74:78: error: invalid use of non-static data member ‘address_range::type’
 ess_range(MAIN_RAM_BANKED_START, MAIN_RAM_BANKED_END, address_range::type::NO_CO
```

There also seem to be several other errors with `/pico-sdk/tools/elf2uf2/main.cpp`. Below is a longer build log. Note: that I can successfuly load the pre-compiled blink.uf2 file from the raspberry pi site.It's this build process that's preventing me from developing in C on the pico.

```terminal
home@computer:~/<truncated>/pico/pico-examples/build/blink$ make
Scanning dependencies of target bs2_default
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] No download step for 'ELF2UF2Build'
[  0%] No update step for 'ELF2UF2Build'
[  0%] No patch step for 'ELF2UF2Build'
[  0%] Performing configure step for 'ELF2UF2Build'
[  0%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  0%] Linking ASM executable bs2_default.elf
[  0%] Built target bs2_default
[  0%] Generating bs2_default.bin
[  0%] Generating bs2_default_padded_checksummed.S
[  0%] Built target bs2_default_padded_checksummed_asm
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /<truncated>/pico/pico-examples/build/elf2uf2
/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:72:58: error: invalid use of non-static data member ‘address_range::type’
     address_range(FLASH_START, FLASH_END, address_range::type::CONTENTS),
                                                          ^
[build notes omitted]

/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:73:64: error: invalid use of non-static data member ‘address_range::type’
     address_range(MAIN_RAM_START, MAIN_RAM_END, address_range::type::NO_CONTENTS
                                                                ^
[build notes omitted]

/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:74:78: error: invalid use of non-static data member ‘address_range::type’
 ess_range(MAIN_RAM_BANKED_START, MAIN_RAM_BANKED_END, address_range::type::NO_CO
                                                                      ^
[build notes omitted]

/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:75:1: error: no matching function for call to ‘std::vector<address_range>::vector(<brace-enclosed initializer list>)’
 };
 ^

[build notes omitted]

/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:78:64: error: invalid use of non-static data member ‘address_range::type’
     address_range(MAIN_RAM_START, MAIN_RAM_END, address_range::type::CONTENTS),
                                                                ^
[build notes omitted]

/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:79:64: error: invalid use of non-static data member ‘address_range::type’
     address_range(XIP_SRAM_START, XIP_SRAM_END, address_range::type::CONTENTS),
                                                                ^
[build notes omitted]

/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:80:60: error: invalid use of non-static data member ‘address_range::type’
     address_range(0x00000000u, 0x00004000u, address_range::type::IGNORE) // for 
                                                            ^
[build notes omitted]

/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:81:1: error: no matching function for call to ‘std::vector<address_range>::vector(<brace-enclosed initializer list>)’
 };
 ^

/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:159:51: error: ‘address_range::type’ is not a class, namespace, or enumeration
                     if (ar.type != address_range::type::CONTENTS) {
                                                   ^
/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp: In function ‘bool is_address_initialized(const address_ranges&, uint32_t)’:
/<truncated>/pico/pico-sdk/tools/elf2uf2/main.cpp:224:35: error: ‘address_range::type’ is not a class, namespace, or enumeration
             return address_range::type::CONTENTS == range.type;
                                   ^
CMakeFiles/elf2uf2.dir/build.make:75: recipe for target 'CMakeFiles/elf2uf2.dir/main.cpp.o' failed
make[5]: *** [CMakeFiles/elf2uf2.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/elf2uf2.dir/all' failed
make[4]: *** [CMakeFiles/elf2uf2.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make[3]: *** [all] Error 2
blink/CMakeFiles/ELF2UF2Build.dir/build.make:85: recipe for target 'blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-build' failed
make[2]: *** [blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-build] Error 2
CMakeFiles/Makefile2:3722: recipe for target 'blink/CMakeFiles/ELF2UF2Build.dir/all' failed
make[1]: *** [blink/CMakeFiles/ELF2UF2Build.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2
```

lurch, 2021-07-14:
> The C compiler identification is GNU 5.4.0

IIRC we don't support such an old GCC version.

xayhewalo, 2021-07-14:
@lurch What's the min GCC version required?

lurch, 2021-07-14:
I dunno if there's an "official minimum version" (I guess it's just a case of try it and see if it works!), but ISTR that I've built it at least once using GCC 6.3.0 (although that was a while ago now).

xayhewalo, 2021-07-14:
I'll, update GCC and see what happens

xayhewalo, 2021-07-14:
Upgrading to GCC 7.5.0 fixed the issue.

Gonna close but would the devs be open to a pull request to add a min GCC warning? The build errors don't point to the real problem.

closed 2021-07-14 by xayhewalo