raspberrypi/pico-sdk #511: Using __STATIC_FORCEINLINE coming with cmsis_compiler.h brings extra portability

GorgonMeducer, 2021-07-14:
Hi,

I am starting porting pico-sdk to arm compiler 6, and I notice that:

- pico-sdk uses cmsis
- pico-sdk uses **_\_\_force\_inline\_** (and always together with static)

So, how about use __STATIC_FORCEINLINE provided by CMSIS (and it supports GCC, LLVM, IAR, AC5 and AC6) to gain better portability? 

or maybe you don't have to change each of the **_\_\_force\_inline\_** into **\_\_STATIC\_FORCEINLINE** but simply define it as **\_\_STATIC\_FORCEINLINE** in platform.h

It is just a suggestion.

lurch, 2021-07-14:
IIRC pico-sdk _optionally_ uses cmsis?

GorgonMeducer, 2021-07-14:
I am not sure. But use it could bring some cross-platform benefits.

kilograham, 2021-07-19:
Yeah we won't be using the CMSIS headers directly.

closed 2021-07-19 by kilograham