raspberrypi/pico-sdk #515: pico-sdk boot troubleshooting

GorgonMeducer, 2021-07-15:
Hi, 

I am trying to port pico-sdk to arm compile 6. I manage to do the stage2 boot, but the whole blink demo project doesn't work. 
I am trying to do some troubleshooting while waiting for my new pico as a probe. What I have done now is:

- Simulate the generated code on MPS2 Cortex-M0plus
    * make sure the stage2 boot work with single steps (**DONE**). In this test, XIP initialisation code is temporarily removed 
    comment out after line https://github.com/raspberrypi/pico-sdk/blob/bfcbefafc5d2a210551a4d9d80b4303d4ae0adf7/src/rp2_common/boot_stage2/boot2_w25q080.S#L104
    and before line https://github.com/raspberrypi/pico-sdk/blob/bfcbefafc5d2a210551a4d9d80b4303d4ae0adf7/src/rp2_common/boot_stage2/boot2_w25q080.S#L275

    * make sure the code can enter main (**DONE**). In this test, the MPS2 FPGA Cortex-M0plus platform clearly shows that the processor can enter main.
    * Since I do not fully understand the mechanism behind the sleep_ms(), I replace it with a proofed delay_us() function. But the blink LED demo doesn't work.
    https://github.com/GorgonMeducer/Pico_Template/blob/add-stage2-boot/main.c

Since gpio_init(), gpio_set_dir() and gpio_put() access peripheral registers directly, there could be something wrong with clock configuration and peripheral clock gating (PCLK). So I have several questions:

- Who is responsible to enable the PCLK of GPIO? I have called the runtime_init() immediately after reset.
- Should runtime_init() should always be called after the initialisation of RW and ZI?
- What's the purpose of the binary_info_header, it is required?

Thank you in advance.

GorgonMeducer, 2021-07-15:
[reproducer.zip](https://github.com/raspberrypi/pico-sdk/files/6826667/reproducer.zip)
Here is my image + map + uf2. 
If you get some insight into my problem, that will be much appreciated.

kilograham, 2021-07-15:
I'm still very confused why as part of " port pico-sdk to arm compile 6" you are also throwing out the CMake build system. these are entirely orthogonal

GorgonMeducer, 2021-07-16:
> I'm still very confused why as part of " port pico-sdk to arm compile 6" you are also throwing out the CMake build system. these are entirely orthogonal

Yeah, sorry for this confusion. Let me correct it, I am trying to compile pico-sdk for some static configuration using armclang. It is not legitimate to say I am porting.

GorgonMeducer, 2021-07-16:
Hi, 

Thanks to one of my colleagues, I can now debug Pico with DSTREAM using Arm-DS. 
And I have solved the problem. Now anyone can compile pico-sdk with Arm Compiler 6 (You can use EVAL license since the code size is smaller than 32K)

Program Size: Code=8814 RO-data=690 RW-data=252 ZI-data=39732  

https://github.com/GorgonMeducer/Pico_Template/releases/tag/v0.9.0

Thank you for all your support.

Cheers,
Gabriel

closed 2021-07-16 by GorgonMeducer