raspberrypi/pico-sdk #518: need help | wont build, spent days

Geekume, 2021-07-16:
So far Pico sadly has brought me more pain than happiness. I have been getting an error where the toolchain just straight up doesn't work. and the sad part is it once did. and now it doesnt. 

[build] C:\Users\tjsun\Downloads\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:74:68: error: 'uintptr_t' undeclared (first use in this function); did you mean 'uint64_t'?
[build]    74 | #define hw_xor_alias_untyped(addr) ((void *)(REG_ALIAS_XOR_BITS | (uintptr_t)(addr)))
[build]       |                                                                    ^~~~~~~~~
[build] C:\Users\tjsun\Downloads\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:108:19: note: in expansion of macro 'hw_xor_alias_untyped'
[build]   108 |     *(io_rw_32 *) hw_xor_alias_untyped((volatile void *) addr) = mask;
[build]       |                   ^~~~~~~~~~~~~~~~~~~~
[build] In file included from C:\Users\tjsun\Downloads\pico-sdk\src\common\pico_stdlib\include/pico/stdlib.h:14,
[build]                  from C:\Users\tjsun\Desktop\folder\main.c:2:
[build] C:\Users\tjsun\Downloads\pico-sdk\src\rp2_common\hardware_uart\include/hardware/uart.h: In function 'uart_get_instance':
[build] C:\Users\tjsun\Downloads\pico-sdk\src\rp2_common\hardware_uart\include/hardware/uart.h:105:5: warning: implicit declaration of function 'static_assert'; did you mean 'hard_assert'? [-Wimplicit-function-declaration]
[build]   105 |     static_assert(NUM_UARTS == 2, "");
[build]       |     ^~~~~~~~~~~~~
[build]       |     hard_assert
[build] NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102020~1\bin\AR19DD~1.EXE' : return code '0x1'
[build] Stop.
[build] NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
[build] Stop.
[build] NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
[build] Stop.
[build] Build finished with exit code 2

this is just some of the errors that go on. I have been trying to get help from discord servers and more and yet i still couldn't get it figured out. Probably spent in total over a days worth of hours trying to figure it out but i just cant. if anyone at all knows whats going on I would love and appreciate the enlightenment. 

- Geekume

lurch, 2021-07-16:
Have you tried making a backup of any files you've hand-edited, uninstalling things, and then following the instructions in https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf again?

Sorry that's a bit of a blunt approach, but I don't know how to diagnose what exactly might have gone wrong with your setup. Let us know If it still doesn't work, and hopefully someone else in the team might have some ideas.

Geekume, 2021-07-16:
Yes I had uninstalled things and reinstalled things. The code is the way it originally was unless I was supposed to touch it but usually you aren't supposed to touch sdk files. I have followed the tutorial of doing it all over again still won't work. I've tried uninstalling cmake reinstalling cmake getting a different version of cmake. Uninstalled the arm gnu tool chain. Reinstalled it. Making sure I didn't miss environment variables or if something changed with in vs. I've done practically everything under the sun it seems in terms of reinstalling and uninstalling

lurch, 2021-07-16:
Have you also tried deleting the `pico-sdk` folder and re-cloning it, just in case something's accidentally got corrupted in there?
Sorry that you're having so many problems :confused:

Geekume, 2021-07-16:
Yeah I actually deleted and download the newest version. And it's ok I had an awesome project I wanted to do with pico but the sdk hasn't been working. The weird part too was even when i tried it on Arduino and get it set up with the official plugin for rp pico boards I was having similar issues. Weird question is there configs on the board itself that stays on there. It's just weird that the issue carries over when the plugin has their own arm tools

Geekume, 2021-07-16:
Do any of you got any suggestions?

lurch, 2021-07-16:
Nothing from me I'm afraid (I'm working from home), but "hopefully someone else in the team might have some ideas." :shrug: 
And if you've not already done so, try asking for help at https://www.raspberrypi.org/forums

aallan, 2021-07-16:
Have you tried the third party Windows installer?  https://github.com/ndabas/pico-setup-windows

Geekume, 2021-07-16:
I haven't but I guess I'll try that