raspberrypi/pico-sdk #531: filesystem support

kilograham, 2021-07-27:
maybe add some minimal stubs for common libraries like littlefs (i.e. make them trivial to use if the littlefs code is present)

aallan, 2021-07-28:
If it gets to the top of your pile soon enough, maybe we can get one of this year's interns to do some SD Card examples with littlefs for pico-examples?

speedypotato, 2021-08-01:
Glad to see this is on the radar, and +1 on examples of writing a file from host and reading it on the pi pico.

deanhuff, 2021-08-03:
i would very much like support for this

rmeshpresh, 2022-06-30:
Hello, please add support for this (sd card on spi) as it is basic requirements for many commercial products. We are only this support short of using rp2040. Please support this asap.

robotman2412, 2023-02-27:
I am also running into this issue.
I've been trying to analyse the SDK but find no definitions of `open` nor `fopen` (somewhat unsurprisingly).
Is there any realistic way to "hack" this in?
My efforts of searching for the connection between `f...()` stdio functions and SDK implementations hasn't yet yielded any results.

rmeshpresh, 2023-02-28:
I think they are not even made example for spi sd card. Even basic
microcontroller provide this support out of box in their sdk.