raspberrypi/pico-sdk #539: Can no longer compare `absolute_time_t` with comparison operator?

infovore, 2021-08-20:
I had working code against a pre-1.2.0 SDK that compared two `absolute_time_t` using `<`, eg:

```
if(storedAbsTimeT < get_absolute_time()) {
```

This now throws an error:

> `no operator "<" matches these operands -- operand types are: absolute_time_t < absolute_time_t`

Was my code (which seemed to work correctly) erroneous before; is this a regression - or should I not be using absolute times in this way?

lurch, 2021-08-20:
I think whether you can use `<` or not depends on whether you're using a Debug or Release build, see #255 
But according to https://raspberrypi.github.io/pico-sdk-doxygen/group__timestamp.html `absolute_time_t` is an "opaque type", so in order to directly compare two values I guess you'd need to use [absolute_time_diff_us](https://raspberrypi.github.io/pico-sdk-doxygen/group__timestamp.html#ga36d9cbebcc55e41921550edbabf17ab2). Or you could use [to_us_since_boot](https://raspberrypi.github.io/pico-sdk-doxygen/group__timestamp.html#ga173698876ae00331c92ed4e8c3ba98d1) if you wanted to do the comparison more directly?

infovore, 2021-08-21:
Yeah, I've moved to using `absolute_time_diff_us`. Like I said, not sure if this was behaviour that shouldn't have been possible that I used in error, or if it's a regression - happy to use the comparison function with opaque types.

lurch, 2021-08-31:
@kilograham I've just had a thought - given that [absolute_time_diff_us](https://raspberrypi.github.io/pico-sdk-doxygen/group__timestamp.html#ga36d9cbebcc55e41921550edbabf17ab2) warns about "be careful when diffing against large timestamps (e.g. at_the_end_of_time) as the signed integer may overflow.", could adding a `absolute_time_cmp` function that returns -1, 0 or 1 (i.e. similar to `strcmp`) be useful?