raspberrypi/pico-sdk #551: 'nosys.specs': No such file or directory

EXtremeExploit, 2021-08-31:
Output of make:
```sh
❯ make
[  1%] Performing build step for 'ELF2UF2Build'
Consolidate compiler generated dependencies of target elf2uf2
[100%] Built target elf2uf2
[  3%] No install step for 'ELF2UF2Build'
[  4%] Completed 'ELF2UF2Build'
[ 13%] Built target ELF2UF2Build
[ 14%] Linking ASM executable bs2_default.elf
arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory
compilation terminated.
make[2]: *** [pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/build.make:97: pico-sdk/src/rp2_common/boot_stage2/bs2_default.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1415: pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

PICO_SDK_PATH is set correctly and standards are
CMAKE_C_STANDARD 11
CMAKE_CXX_STANDARD 17

already went to pico-sdk folder and did git submodule update --init --recursive
idk what else to do

OS: Arch Linux

kilograham, 2021-08-31:
most likely you have a weird version of ARM GCC (maybe arch linux releated).

Where did you get it/how did you install, dd you follow the getting started instructions.

clean `cmake` output might be helpful, and `arm-non-eabi-gcc --version` output

EXtremeExploit, 2021-08-31:
i got it from the official arch repositories, output:
```sh
❯ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

and clean cmake output:
```sh
❯ cmake ..
Using PICO_SDK_PATH from environment ('/home/pedro/Documentos/pico-sdk')
PICO_SDK_PATH is /home/pedro/Documentos/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/pedro/Documentos/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.9 (found version "3.9.6") found components: Interpreter
TinyUSB available at /home/pedro/Documentos/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pedro/Documentos/dev/kappa-pad/build
```

kilograham, 2021-08-31:
GCC 11 is not necessarily known to work, although that probably isn't the problem.
I suspect the arch linux version does not include `newlib` ... you can probably figure that out, but maybe simpler to install compiler from here https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm

EXtremeExploit, 2021-08-31:
exactly, you were right, it doesn't come with newlib, had to install arm-none-eabi-newlib from arch repo, this solved the issue, thanks :D

closed 2021-08-31 by EXtremeExploit