raspberrypi/pico-sdk #552: how to use rename_exceptions.h

xiongyu0523, 2021-09-02:
By default this file does not take effects, in final image i can't find replaced symbol., though default value of PICO_CMSIS_RENAME_EXCEPTIONS is 1. 

At higher level I need LIB_CMSIS_CORE to be 1, but I can't find any reference to this macro in the code.

In \src\rp2_common\cmsis\CMakeLists.txt, looks like PICO_CMSIS_CORE_PATH is defined by default.

xiongyu0523, 2021-09-02:
add cmsis_core to my project's link libraries I can see system_RP2040.c is build but I can't Systick_Hanlder stuff in elf. How can I let rename_expections.h take effect?

lurch, 2021-09-02:
Appendix F in https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf mentions:

**Added CMSIS core headers**
CMSIS core headers (e.g. `core_cm0plus.h` and `RP2040.h`) are made available via `cmsis_core` INTERFACE library. Additionally, CMSIS standard exception naming is available via `PICO_CMSIS_RENAME_EXCEPTIONS=1`

xiongyu0523, 2021-09-03:
Thanks. Is there any example in pico-example has a demonstration of using this cmsis_core INTERFACE library?

lurch, 2021-09-03:
Not yet, but we welcome PRs... :wink: