raspberrypi/pico-sdk #553: How do you use board detection

thezim, 2021-09-03:
I have a project I'd like to detect boards in so I get the I2C pins for use in my header file.

kilograham, 2021-09-03:
Not sure what you mean.

Setting of PICO_BOARD cmake var determine which board you are compiling for and will generally select one of the board headers here https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards

most of those have defines that let you tell what the board is e.g. https://github.com/raspberrypi/pico-sdk/blob/bfcbefafc5d2a210551a4d9d80b4303d4ae0adf7/src/boards/include/boards/pico.h#L18 however for simple stuff the board header already defines vars like `PICO_DEFAULT_I2C_SDA_PIN`

thezim, 2021-09-03:
That was exactly what I was looking for. Thanks.

closed 2021-09-03 by thezim

lurch, 2021-09-09:
For the benefit of anybody reading this issue in the future, see section "2.6.1. Preprocessor Variables via Board Configuration File" in https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf