raspberrypi/pico-sdk #558: Printing double precision

ejolson2005, 2021-09-11:
Adding

`pico_set_printf_implementation(TARGET compiler)`

to CMakeLists.txt leads to the error

`/x/liba/local/pico-sdk/src/rp2_common/pico_runtime/runtime.c:23:10: fatal error: pico/printf.h: No such file or directory
   23 | #include "pico/printf.h"`

Expected result is compilation using the compiler implementation of printf that properly prints double precision variables.

Suggested work around (untested) is

`#define weak_raw_printf printf
#define weak_raw_vprintf vprintf`

as stated in 

https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=319434

see also the work around

`target_link_libraries(TARGET pico_stdlib pico_printf_headers)`

as suggested in

https://www.raspberrypi.org/forums/viewtopic.php?p=1849648#p1849648

Thanks for looking at this.

ejolson2005, 2021-09-12:
It would appear either work around causes the output from consecutive calls to __wrap_printf and __wrap_putchar among others to appear out of order.  Note that the compiler can generate a __wrap_putchar for something like printf("\n").

kilograham, 2021-10-12:
merged into develop

closed 2021-10-12 by kilograham