raspberrypi/pico-sdk #567: i got a issue about uart,it could not receive any data

raffeale, 2021-09-19:
hi,i'm a new guy for pico,i use arduino to develop some project, recently i used pico with a uart wifi module to send data to other mcu,the other mcu use wifi module as well. when i use uart wifi module on pico ,some time i could only get correct data once, it usually get wrong data or 0 from wifi .i'm pretty sure that it's pico problem,because the same code could work on mega328p or esp32. 
even if i used hardware uart api , i only get 0 numbe from uart.

lurch, 2021-09-20:
This sounds like a question for the [forums](https://www.raspberrypi.org/forums/viewforum.php?f=143) ?

raffeale, 2021-09-20:
today i have tested the waveform on uart rx, i got correct data on my oscilloscope , the waveform is right but pico didn't get any correct data, some time it's always 0, and i notice that the rx will pull down the rx voltage from 3.3v to 1.5v

raffeale, 2021-09-20:
i use uart hardware api to get data ,it get 0. and i read uartrsr register i found it is 8 ,that means fifo is overrrun, sometime rsr is 13 i did not use uart parity , my uart mode is N81

aallan, 2021-09-20:
This is the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/viewforum.php?f=145), so I’m closing this issue. But please follow up with further comments if you have an issue with the SDK itself.

closed 2021-09-20 by aallan

lurch, 2021-09-20:
This _might_ be related to #548 ?