raspberrypi/pico-sdk #569: API to support generate random number?

xiongyu0523, 2021-09-21:
is there a plan to add hardware random number generation?

lurch, 2021-09-21:
Are you looking for "2.17.5. Random Number Generator" in https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf ?
Or are you asking for a "nice API" to wrap that?

Wren6991, 2021-09-21:
You can read the ring oscillator random bit with something like

```c
#include "hardware/structs.rosc.h"

bool get_random_bit() {
    return rosc_hw->randombit;
}
```

This is just an unconditioned ring oscillator sample. It's not well characterised, somewhat biased, output is somewhat periodic when sampled rapidly, etc. We haven't wrapped it up in a simplified API because that might give an impression of randomness guarantees that we haven't done the characterisation to back up. It's very much "some assembly required".

What functionality are you looking for exactly?

xiongyu0523, 2021-09-24:
Thank you very much for your example and forum discussions. I will use the rand number for the SSL library port. I can understand the hardware characteristic is not defined yet, but providing such a API is still useful and can be a seed feed to srand

makeworld-the-better-one, 2022-03-21:
This is fixed thanks to #4213. You can generate randomness on the RP2040 using `os.urandom` .

lurch, 2022-03-21:
@makeworld-the-better-one It sounds like you might be mixing up the RP2040 C SDK with the RP2040 MicroPython port?

makeworld-the-better-one, 2022-03-21:
Yep sorry, my bad

kilograham, 2022-07-14:
This seems like a useful addition, as the random bit is not sufficiently random if polled at high frequency

cameronelliott, 2022-09-02:
> Thank you very much for your example and forum discussions. I will use the rand number for the SSL library port. I can understand the hardware characteristic is not defined yet, but providing such a API is still useful and can be a seed feed to srand

You did see the note in the forums? "The datsheet does say that the randomness is not suitable for cryptography, since it can be compromised, and thus is only suitable for less critical applications."

kilograham, 2023-01-20:
fixed by #1111

kilograham, 2023-01-27:
merged into develop

closed 2023-01-27 by kilograham