raspberrypi/pico-sdk #579: i2c_write* problem addressing standard i2c 24cxx eeprom size > 2Kbit

louigi600, 2021-10-02:
The 24cxx family i2c eeproms expect to receive the address of where you want to write before the actual data you want to wrote.
So we should be writing to i2c  2 elements  (via src): address, data 
Now if the address can only be uint8_t 

int i2c_write_blocking (i2c_inst_t *i2c,
uint8_t addr,
const uint8_t *src,
size_t len,
bool nostop)

how am I supposed to address beyond 255 ?
I suspect that we should be allowed to have something bigger in 
const uint8_t *src
else what us the point of 24c04, 24c08 and 24c16 (expectively 515byte, 1kbyte, 2kbyte) if we can only address the first 255 bytes anyway ?

Maybe we are supposed to send address in restart mode in 2 parts ? (by setting nostop to trwue while writing the address)
but the datasheet clearly states that the address is expected to be a byte ....
Maybe I'm missing something ... I looked for specific c/c++ pico sdk examples of reading and writing i2c eeprom, but I found none.

Besides this problem on the addressing I'm having issues rereading back after I write data (even limited to the first 256 bytes) ...
I need to reset the eeprom to get a consistent read of what I wrote.

lurch, 2021-10-02:
I think you might be misunderstanding I2C? IIRC the `uint8_t addr` is the address of the I2C _device_ on the bus, as multiple I2C devices can be connected to the same bus, each with a different device address.
I think the address of the data that you want to read from/write to the device would be part of the `const uint8_t *src` structure? (and you'll need to check the datasheet of your device to find out the exact protocol that it uses)

louigi600, 2021-10-02:
No I'm talking about const uint8_t *src
I'm talking about addressing the data inside an i2c eeprom .

I solved the reading back consistency: it was my fault I was not giving the memory enough time to complete the write operation.
The rest still holds true: I have not figured out how to address the data inside the eeprom beyond byte 255.

lurch, 2021-10-02:
"you'll need to check the datasheet of your device to find out the exact protocol that it uses" e.g. https://www.mouser.co.uk/datasheet/2/268/Atmel-8787-SEEPROM-AT24C04C-08C-Datasheet-1398080.pdf

lurch, 2021-10-02:
You might want to also try asking for help on the Pico forums https://www.raspberrypi.org/forums/viewforum.php?f=143

louigi600, 2021-10-02:
> "you'll need to check the datasheet of your device to find out the exact protocol that it uses" e.g. https://www.mouser.co.uk/datasheet/2/268/Atmel-8787-SEEPROM-AT24C04C-08C-Datasheet-1398080.pdf

Yes datasheet says:

> Page Write. For the Page Write mode, the MODEpin must be at VIL. The Page Write mode allows upto  8  bytes  to  be  written  in  a  single    write  cycle,provided that they are all located in the same ’row’in the memory: that is the 5 most significant mem-ory address bits (A7-A3) are the same. The mastersends from one up to 8 bytes of data, which areeach  acknowledged  by  the  memory.  After  eachbyte is transfered, the internal byte address counter(3  least  significant  bits  only)  is  incremented.  Thetransfer is terminated by the master generating aSTOP condition. Care must be taken to avoid ad-dress counter ’roll-over’ which could result in databeing  overwritten.  Note  that,  for  any  write  mode,the generation by the master of the STOP conditionstarts the internal memory program cycle. All inputsare disabled until the completion of this cycle andthe memory will not respond to any request.

Now the problem is how to I writhe the address I want to reach if it is > then uint_8t ?
it hasto fit into an element of src ... but they are uint8_t as per i2c_write_blocking definition

lurch, 2021-10-02:
I've never written any EEPROM code myself, but it looks like this might be relevant?
> The 4K EEPROM only uses the A2 and A1 device address bits with the third bit being a memory page address
bit. The two device address bits must compare to their corresponding hard-wired input pins. The A0 pin is no
connect. 
The 8K EEPROM only uses the A2 device address bit with the next two bits being for memory page addressing.
The A2 address bit must compare to its corresponding hard-wired input pin. The A1 and A0 pins are no connect.

louigi600, 2021-10-02:
So protocol is same but I will need to wire the 3 extra address pins to addtess upto 8 * 256 segments. 
I thought all the address pins were to solve conflicting address issues on the bus ... this may only hold true on 24c02.
But the 24c64 can't work the same way protocol must be different because even the full 3 extra addtess bits will not cover the full  8Kb.
Anyway the code I showed now works fine up-to the first 255 bytes.

lurch, 2021-10-02:
> But the 24c64 can't work the same way protocol must be different because even the full 3 extra addtess bits will not cover the full 8Kb.

Yes, https://www.mouser.co.uk/datasheet/2/389/cd00259166-1797197.pdf says that there's both a "Most significant address byte" and a "Least significant address byte".

louigi600, 2021-10-02:
@lurch : maybe you are the right guy to tell the people maintaining pico-example that a i2c eeprom example would be nice.
I think this issue never really existed ... but it worked better then the forum.

closed 2021-10-02 by louigi600

beta-tester, 2023-10-26:
hi, i know it is a very old issue, but i was stumbling around the same issue.
i found how addressing works beyond the 0xFF memory address in the arduino library [i2c_eeprom](https://github.com/RobTillaart/I2C_EEPROM).
the addressing beyond memory address 0xFF is done by
- addressing as @lurch said (transmit high byte of the memory address, then low byte of the memory address)
  and [see here in i2c_eeprom library](https://github.com/RobTillaart/I2C_EEPROM/blob/1.7.4/I2C_eeprom.cpp#L494)
- or use page write mode (translate addresses in pages and write data in page size chunks / page size depends on the eeprom you use). [see here in i2c_eeprom library](https://github.com/RobTillaart/I2C_EEPROM/blob/1.7.4/I2C_eeprom.cpp#L470)