raspberrypi/pico-sdk #614: alarm_pool_add_alarm_at() returns 0

smdjeff, 2021-10-15:
I believe there is some confusion about whether or not alarm_id 0 is valid.  I tried this and it fails...
```
bool success = add_repeating_timer_ms( millis, callback, userdata, &timers ); 
assert( success );
```
Perhaps the Timer documentation and alarm_pool_add_repeating_timer_us() return calculation is incorrect?

kilograham, 2021-10-15:
I assume you mean the `return out->alarm_id > 0;` in add_repeating_time_ms.
yes that looks like a bug; should be `>= 0`

smdjeff, 2021-10-19:
> alarm_id_t
> typedef int32_t alarm_id_t
> The identifier for an alarm.
> 
> Note
> this identifier is signed because -1 is used as an error condition when creating alarms
> alarm ids may be reused, however for convenience the implementation makes an attempt to defer reusing as long as possible. You should certainly expect it to be hundreds of ids before one is reused, although in most cases it is more. Nonetheless care must still be taken when cancelling alarms or other functionality based on alarms when the alarm may have expired, as eventually the alarm id may be reused for another alarm.

https://raspberrypi.github.io/pico-sdk-doxygen/group__alarm.html

and several api say...

> Returns
> >0 the alarm id
> 0 the target timestamp was during or before this method call (whether the callback was called depends on fire_if_past)
> -1 if there were no alarm slots available

really... i think there's a lot of inconsistency in the api return values. they should probably all return traditional error codes. with 0 = success. rather than mixing and matching ids and success = true booleans.  maybe this is a hold over from the python world rather than taking a shot at if from the *nix world.

smdjeff, 2021-10-19:
```
alarm_pool_add_alarm_at()
        ...
       public_id = missed ? 0 : make_public_id(id_high, id);
        if (missed && fire_if_past) {
            int64_t repeat = callback(public_id, user_data);
            if (!repeat) {
                public_id = 0;
                ...
        return public_id;
```
I suppose that means alarm_id = 0 is a special case that means "calledback because missed, but not repeated" ?