raspberrypi/pico-sdk #645: fatal error: cstdlib: No such file or directory

Nippius, 2021-11-01:
Hi 

I've updated pico-sdk to the latest version (1.2.0) and I'm getting an error regarding missing **cstdlib** when compiling projects.

I can replicate this by compiling **hello_usb** from the pico-examples repo:

```
make hello_usb -j
[  0%] Built target bs2_default
[  0%] Performing build step for 'ELF2UF2Build'
[  0%] Built target bs2_default_padded_checksummed_asm
[100%] Built target elf2uf2
[  0%] No install step for 'ELF2UF2Build'
[  0%] Completed 'ELF2UF2Build'
[  0%] Built target ELF2UF2Build
Consolidate compiler generated dependencies of target hello_usb
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/src/rp2_common/pico_standard_link/binary_info.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/src/rp2_common/pico_stdio_usb/reset_interface.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/src/rp2_common/pico_stdio/stdio.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/src/rp2_common/pico_stdio_usb/stdio_usb.c.obj
[  0%] Building CXX object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/src/rp2_common/pico_stdio_usb/stdio_usb_descriptors.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/device/usbd.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/device/usbd_control.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/class/audio/audio_device.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/class/cdc/cdc_device.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/class/dfu/dfu_device.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/class/dfu/dfu_rt_device.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/class/net/net_device.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/class/midi/midi_device.c.obj
[  0%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c.obj
/home/xxxx/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:10:10: fatal error: cstdlib: No such file or directory
   10 | #include <cstdlib>
      |          ^~~~~~~~~
compilation terminated.
make[3]: *** [hello_world/usb/CMakeFiles/hello_usb.dir/build.make:653: hello_world/usb/CMakeFiles/hello_usb.dir/home/xxxx/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:3804: hello_world/usb/CMakeFiles/hello_usb.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:3811: hello_world/usb/CMakeFiles/hello_usb.dir/rule] Error 2
make: *** [Makefile:202: hello_usb] Error 2
```

Here's are the libraries I have installed:
```
sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
cmake is already the newest version (3.21.3-4).
gcc-arm-none-eabi is already the newest version (15:10.3-2021.07-1).
libnewlib-arm-none-eabi is already the newest version (3.3.0-1).
libstdc++-arm-none-eabi-newlib is already the newest version (15:8-2019-q3-1+13).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Also, I'm running Kali Linux on WSL2:

```
uname -a
Linux OKLO 5.10.16.3-microsoft-standard-WSL2 #1 SMP Fri Apr 2 22:23:49 UTC 2021 x86_64 GNU/Linux
```

Any idea what might be causing this?

Thanks!

lurch, 2021-11-02:
I don't think any of the Raspberry Pi engineers working on Pico are using "Kali Linux on WSL2", so you _might_ get a quicker response by asking for help on the forums? https://forums.raspberrypi.com/viewforum.php?f=143

kilograham, 2021-11-02:
I mean it <cstdlib> should be included with GCC or at least G++

kilograham, 2021-11-02:
you can try running `make VERBOSE=1` when you build... and check the command line (though that may be limited as the SDK does _not_ add any paths to the C headers itself)

So failing that maybe try compiling a single cpp file including <cstdio> using the ARM gcc compiler directly

Nippius, 2021-11-02:
> I mean it should be included with GCC or at least G++

My thoughts exactly. Really weird, but this is running on a VM with no other software so I can only assume an update broke something.

Infact I just tried on a ubuntu machine and it's working fine. 

> So failing that maybe try compiling a single cpp file including using the ARM gcc compiler directly
Unfortunately that also fails with the same error and no other info.

I think I'll leave it as is for now, I'll ask around in the forums like @lurch mentioned. In the meantime I'll use Ubuntu as thats working as expected.

Thanks!

Nippius, 2021-11-02:
Well I've just checked the forum and it looks like I'm not the only one with this problem: https://forums.raspberrypi.com/viewtopic.php?t=322624

alexm622, 2021-11-08:
> Well I just checked the forum and it looks like I'm not the only one with this problem: https://forums.raspberrypi.com/viewtopic.php?t=322624

I also have this issue, I am running Kali on baremetal and I get the exact same error, at the exact same spot.

kilograham, 2021-11-11:
this is not a SDK issue - an OS problem (see forum post)

closed 2021-11-11 by kilograham

raviatnunam-01, 2021-12-17:
sudo apt install libstdc++-arm-none-eabi-newlib

This command worked for me.