raspberrypi/pico-sdk #649: SDK 1.3.0, CMake, and TinyUSB HOST_HID - OK compile but operation is wrong

dshadoff, 2021-11-02:
I believe that something in the cmake hierarchy is overriding something important in TinyUSB's HOST_HID stack, leading to a runtime issue.

I had a project based on SDK 1.2.0 and TinyUSB's host_cdc_msc_hid example (any version from July 4 onward), which was working fine.  Compiling with the new SDK 1.3.0 compiles fine, but no USB mouse messages reach the code.

Putting that aside, I decided to recheck that the TinyUSB example was still working properly:
1) From inside the ($PICO_SDK)/lib/tinyusb/,  I ran 'make BOARD=raspberry_pi_pico all', and tested the compiled .uf2 file, which worked perfectly.
2) I then tried the corresponding pico_examples build, which appears in $(pico_examples)/build/usb/host/host_cdc_msc_hid  after the full tree build.  The output (using the same hardware) was:
```
TinyUSB Host CDC MSC HID Example
assertion "ep->active" failed: file "/home/pi/devel/pico/pico-setup/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line 164, function: hw_trans_complete

```

I would file this as an issue under pico_examples, but it also affects my project which is separate and not dependent on pico-examples.

kilograham, 2021-11-02:
is 1 vs 2 possibly a release build vs debug build thing?

dshadoff, 2021-11-02:
I built as defaults; are you saying that the default build in pico-examples is debug ?
(I don't receive an assert on my (non-debug) build of my own project, but failing to react to mouse events is similar behaviour other than the lack of assert output)

Difference in operation between 1 and 2 is that displays mouse events, and 2 simply stops doing anything after the assert.

kilograham, 2021-11-02:
debug is not the default (well from the command line - might be in your IDE), but `assert` should do nothing in a release build. but yeah i was wondering if the only reason 1 does something is that the assertion is suppressed.

dshadoff, 2021-11-02:
For context, I develop for pico on a Raspberry Pi 400 from command line in shell.  To try out the new sdk 1.3.0, I started with a fresh git clone and run of pico-setup (after moving the top-level of the previous one aside by renaming it).

kilograham, 2021-11-02:
OK - well i see the same issue with TinyUSB HID mode with a debug build, but not with a release build. With a release build mouse works fine for me.

Seems like TinyUSB have broken the debug version.

I didn't realize pico-setup defaults the build to Debug. Simplest thing is to create a clean build directory and run `cmake ..` again

lurch, 2021-11-03:
Or you can create a clean build directory and run either `cmake .. -DCMAKE_BUILD_TYPE=Release` or `cmake .. -DCMAKE_BUILD_TYPE=Debug` to explicitly force the build-type.

dshadoff, 2021-11-03:
Yes, this was the chain of events:
- default build of pico-examples from pico-setup is a debug build
- TinyUSB has some sort of issue on debug build

After this was shown to be the case, I examined changes to the Host HID example code, and it turns out that several of the axiomatic defines variables were changed in the intervening time (not just values, but also names !).  So I have fixed my problem.

I feel like this issue should probably remain open until the TinyUSB issue on debug builds is fixed, however.

kilograham, 2021-11-17:
we will pick up the fixed upstream TinyUSB in 1.3.1

kilograham, 2022-03-22:
@liamfraser to decide what to include for 1.3.1 or wait for 1.3.2

kilograham, 2022-05-10:
Moving to 1.4.0 as we will not pick up new TinyUSB in 1.3.x branch

kilograham, 2023-02-06:
merged into develop

closed 2023-02-06 by kilograham