raspberrypi/pico-sdk #653: TinyUSB Audio is flawed

TheMindVirus, 2021-11-11:
It has been bugging myself and all the people who want RP2040-based boards to be used as a USB Synthesizer or Audio Unit.

It seems like you are in need of some basic definitions:
Synthesizer - any machine which produces audio data (doesn't need heavy libraries for that, only arrays and basic circular functions)
Audio Unit - any machine which inputs audio data, processes it and outputs the result of the process (e.g. 2ch in 2ch out Reverb unit)
Sequencer - any machine which outputs a timed pulse on a loop for triggering Synthesizers and remote controlling Audio Units
DAW - Digital Audio Workstation, any software which collects Synthesizers, Audio Units and Sequencers for Audio Editing
The same principles may apply to video; instead of the UAC2 specification, the UVC specification (wherever it is) is used instead.

This is the specific line that it has taken a long while to find and that may have been bugging me:
https://github.com/hathach/tinyusb/blob/fc59515bfd14da74fce7e8bd4390586ac8ff3b15/examples/device/audio_test/src/main.c#L4
It seems to create noise characteristic of non-isolated headphone jacks of old laptops, especially audible over a PA System.
It might be there or it could be coming from somewhere else but the source is definitely firmware related.

Since GitHub doesn't support uploading MP3 files, here are the fully digital recordings of what should be Sine and Silence
audible through any RP2040 board:
[PicoSample.zip](https://github.com/raspberrypi/pico-sdk/files/7517554/PicoSample.zip)
The same issue is affecting my 2i2o or 2120 firmware called "Pico Audio Unit", effectively acting as an unintentional bit-crusher.
I can't do my job if you can't do yours. Please kindly take a look at it and also the wider impact it had on products already released.

aallan, 2021-11-11:
Is this a [TinyUSB](https://github.com/hathach/tinyusb) issue, rather than an issue with the Pico C SDK? While our C SDK pulls in TinyUSB as a submodule, we don't maintain it.

TheMindVirus, 2021-11-11:
I have seen the pico-sdk change and between versions the examples from tinyusb have been included in the build output in the usb subfolder.

It seems the previous usb audio example I was using as a template got removed in its place and replaced entirely with tinyusb example code in a different directory. There is another version in pico-extras which is the one marked as not ready for public use.

You have a fork at https://github.com/raspberrypi/tinyusb
yet the pico-sdk still points to the fork of the original at https://github.com/hathach/tinyusb
and Adafruit have their own version of the same thing for the Macropad RP2040 at https://github.com/Adafruit/adafruit_tinyusb_arduino

I used both an Adafruit RP2040 and a Standard Pico to record the samples included in the original post, both had identical results and no analog audio cables are part of the setup.

It looks like the "i2s_dummy_data" is causing some high-level digital interference, audible when put through an amplifier. It could be somewhere else that's causing it.

This affects the publicly published articles on how to turn the Pico into a USB Microphone, leaving some users disappointed with the quality.

aallan, 2021-11-11:
> You have a fork at https://github.com/raspberrypi/tinyusb

We're now using the [upstream repo](https://github.com/hathach/tinyusb) within Pico SDK rather than the fork which was made so we could work on USB support in the lead up to launch back in January.

kilograham, 2021-11-11:
If i can parse this issue correctly, it is that the "audio_test" example in TinyUSB isn't working correctly. This is not SDK related; I suggest you open an issue with TinyUSB

closed 2021-11-11 by kilograham

TheMindVirus, 2021-11-11:
Apologies, while the original post had the permalink pointing to audio_test: https://github.com/hathach/tinyusb/blob/fc59515bfd14da74fce7e8bd4390586ac8ff3b15/examples/device/audio_test/src/main.c#L4
It should have in fact read the following:
https://github.com/raspberrypi/tinyusb/blob/d49938d0f5052bce70e55c652b657c0a6a7e84fe/examples/device/audio_4_channel_mic/src/main.c#L419
There were several tabs open trying to find the source of the interference.

aallan, 2021-11-11:
As I mentioned above, https://github.com/raspberrypi/tinyusb is a legacy repo that is *no longer used by us or the SDK*. We now use the upstream repo. If you have a problem with USB you need to file an issue against the upstream repo, https://github.com/hathach/tinyusb. We do not maintain that library.

dhalbert, 2021-11-11:
> As I mentioned above, https://github.com/raspberrypi/tinyusb is a legacy repo that is _no longer used by us or the SDK_.

Perhaps it would be worth marking this repo as "Archived".