raspberrypi/pico-sdk #654: IVT Syscall Support Request/Feature

spit4520, 2021-11-11:
Hi All,

I am looking to play around with triggering software interrupts on the RP2040, and I have gotten some conflicting documentation on weather or not this would be possible. The IVT should be able to be configured from the V6 spec from ARM, but I can't quite determine if that is possible on the 2040.

I want to experiment with "thread safety" and airlift some of the code at my job out of deep gluey MCU code into more isolated processes. I know that RTOS exists, this is more of a learning exercise and I would like to add a few examples to the main repo.

Could someone point me at the correct place in either the ARM docs or the Pico docs on how to manually configure an internal software interrupt? (I am trying to avoid the hack of using the PIO between multiple cores to use that for interrupting)

Thanks!

Scott

lurch, 2021-11-11:
"2.3.2. Interrupts" in https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf and "4.1.14. hardware_irq" in https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf ?

kilograham, 2021-11-17:
this is not an SDK issue (if the documentation is insufficient then open an issue in pico-feedback)

closed 2021-11-17 by kilograham