raspberrypi/pico-sdk #659: Compiling Error: duplicate 'inline'

splch, 2021-11-15:
When running `make`, the SDK fails to build. The error thrown is:

```shell
error: duplicate 'inline'
```
on file [pico-sdk/src/rp2_common/pico_platform/include/pico/platform.h](https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2_common/pico_platform/include/pico/platform.h)

kilograham, 2021-11-15:
what compiler version are you using, what OS?

splch, 2021-11-15:
@kilograham Here is what I'm using.

Compiler: gcc 8.3.0
cmake version 3.13.4

OS: Debian 10

aallan, 2021-11-16:
From raspberrypi/pico-feedback#206 looks like this might be a [project generator](https://github.com/raspberrypi/pico-project-generator) problem. Did you create your project using the generator?

splch, 2021-11-16:
@aallan I actually just cloned and built the pico-sdk normally. I used these commands:
```shell
git clone https://github.com/raspberrypi/pico-sdk.git ~/pico-sdk
mkdir ~/pico-sdk/build/; cd ~/pico-sdk/build/
cmake ..
make
```
I was compiling for the Picosystem, though. Do you think that could be related to the project generator?

lurch, 2021-11-17:
Hmm, I just tried to reproduce this in a Docker container, and was unable to? On my Ubuntu x64 laptop I ran:
```
docker run -it debian:10
```
and then inside the docker container I ran:
```
apt update && apt install -y git python3 g++ cmake gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib
git clone https://github.com/raspberrypi/pico-sdk.git
cd pico-sdk
mkdir build && cd build
```

When I then run `cmake ..` it says:
```
PICO_SDK_PATH is /pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Defaulting PICO target board to pico since not specified.
Using board configuration from /pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.7 (found version "3.7.3") found components:  Interpreter 
CMake Warning at src/rp2_common/tinyusb/CMakeLists.txt:10 (message):
  TinyUSB submodule has not been initialized; USB support will be unavailable

  hint: try 'git submodule update --init' from your SDK directory
  (/pico-sdk).

Including custom CMake file /pico-sdk/src/common/pico_base/generate_config_header.cmake
ELF2UF2 will need to be built
-- Configuring done
-- Generating done
-- Build files have been written to: /pico-sdk/build
```

and running `make` also then runs to completion, with no errors.
Start of the make output looks like:
```Scanning dependencies of target bs2_default
[  0%] Building ASM object src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  0%] Linking ASM executable bs2_default.elf
[  0%] Built target bs2_default
Scanning dependencies of target ELF2UF2Build
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] No download step for 'ELF2UF2Build'
[  0%] No patch step for 'ELF2UF2Build'
[  0%] No update step for 'ELF2UF2Build'
[  1%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /pico-sdk/build/elf2uf2
[  1%] Performing build step for 'ELF2UF2Build'
Scanning dependencies of target elf2uf2
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[  1%] No install step for 'ELF2UF2Build'
[  1%] Completed 'ELF2UF2Build'
[  1%] Built target ELF2UF2Build
Scanning dependencies of target bs2_default_padded_checksummed_asm
[  1%] Generating bs2_default.bin
[  1%] Generating bs2_default_padded_checksummed.S
[  1%] Built target bs2_default_padded_checksummed_asm
Scanning dependencies of target pico_stdlib_test
[  1%] Building C object test/pico_stdlib_test/CMakeFiles/pico_stdlib_test.dir/pico_stdlib_test.c.obj
[  1%] Building C object test/pico_stdlib_test/CMakeFiles/pico_stdlib_test.dir/__/__/src/rp2_common/pico_stdlib/stdlib.c.obj
```
and ends with:
```
[100%] Building C object test/cmsis_test/CMakeFiles/cmsis_test.dir/__/__/src/rp2_common/pico_stdio_uart/stdio_uart.c.obj
[100%] Building C object test/cmsis_test/CMakeFiles/cmsis_test.dir/__/__/src/rp2_common/cmsis/stub/CMSIS/Device/RaspberryPi/RP2040/Source/system_RP2040.c.obj
[100%] Linking CXX executable cmsis_test.elf
[100%] Built target cmsis_test
```

Is there anything "unusual" about your setup @splch , or any ideas why I can't reproduce your problem in a Debian 10 docker container? :man_shrugging:

kilograham, 2021-11-24:
@lurch i think the issue is compiler version, reporter is using gcc 8.3.0 on ubuntu you are using 7.3.1 - maybe try updating the GCC in the docker container to confirm.

Hermann-SW, 2021-11-24:
I saw same error when compiling with updated to latest Raspberry 32bit OS (Buster).
With 1.2 as well as 1.3 SDK, when compiling am Arducam Pico camera demo.
The error message led me to forum posting and that to this issue.
Quite likely that Arducam used Project generator for their example code:
https://github.com/ArduCAM/RPI-Pico-Cam.git
```
pi@raspberrypi400:~/pico.old/RPI-Pico-Cam/tflmicro/build $ time make
...
[  7%] Building CXX object CMakeFiles/pico-tflmicro.dir/src/tensorflow/lite/micro/rp2/debug_log.cpp.obj
In file included from /usr/include/newlib/stdio.h:35:0,
                 from /home/pi/pico.old/RPI-Pico-Cam/tflmicro/src/tensorflow/lite/micro/rp2/debug_log.cpp:40:
/home/pi/pico/pico-sdk/src/rp2_common/pico_platform/include/pico/platform.h:237:31: error: duplicate 'inline'
 #define __force_inline inline __always_inline
                               ^
...
```

```
pi@raspberrypi400:~/pico/RPI-Pico-Cam/tflmicro/build $ time make
...
[  7%] Building CXX object CMakeFiles/pico-tflmicro.dir/src/tensorflow/lite/micro/rp2/debug_log.cpp.obj
In file included from /usr/include/newlib/stdio.h:35:0,
                 from /home/pi/pico/RPI-Pico-Cam/tflmicro/src/tensorflow/lite/micro/rp2/debug_log.cpp:40:
/home/pi/pico/pico-sdk/src/rp2_common/pico_platform/include/pico/platform.h:237:31: error: duplicate 'inline'
 #define __force_inline inline __always_inline
                               ^
...                          
```

Hermann-SW, 2021-11-24:
My gcc version is "gcc (Raspbian 8.3.0-6+rpi1) 8.3.0", obviously __GNUC__ is not 8.

This workaround in pico-sdk allowed to build complete Arducam Pico camera tensorflow demo without issues:
```
diff --git a/src/rp2_common/pico_platform/include/pico/platform.h b/src/rp2_common/pico_platform/include/pico/platform.h
index ee1d360..ad50883 100644
--- a/src/rp2_common/pico_platform/include/pico/platform.h
+++ b/src/rp2_common/pico_platform/include/pico/platform.h
@@ -234,8 +234,6 @@ extern "C" {
  *
  */
 #if defined(__GNUC__) && __GNUC__ <= 7
-#define __force_inline inline __always_inline
-#else
 #define __force_inline __always_inline
 #endif
 
pi@raspberrypi400:~/pico/pico-sdk $
```

kilograham, 2021-11-24:
> My gcc version is "gcc (Raspbian 8.3.0-6+rpi1) 8.3.0", obviously __⁠GNUC__ is not 8.

I think that may be your native gcc version... what does [b]arm-none-eabi-gcc --version[/b] say

Hermann-SW, 2021-11-24:
You are right:
arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

lurch, 2021-11-24:
I've just repeated the previous docker-test using a `debian:11` container (using the same commands as in my previous comment), and that builds totally fine too :confused: 

Package versions in `debian:10`
```
root@5907d3f880ca:/pico-sdk/build# cmake --version
cmake version 3.13.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
root@5907d3f880ca:/pico-sdk/build# gcc --version
gcc (Debian 8.3.0-6) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@5907d3f880ca:/pico-sdk/build# arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Package versions in `debian:11`
```
root@5348a56c1c9c:/pico-sdk/build# cmake --version
cmake version 3.18.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
root@5348a56c1c9c:/pico-sdk/build# gcc --version
gcc (Debian 10.2.1-6) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@5348a56c1c9c:/pico-sdk/build# arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

That's all on x64 though, so maybe something's working differently when building on ARM? I'll try to remember to do those same compile-tests on Raspberry Pi OS Buster and Raspberry Pi OS Bullseye tomorrow.

kilograham, 2021-11-25:
yup, arm-none-eabi-gcc 7.3.1. and 8.3.1 (amongst many others) build fine for me (albeit on Ubuntu) 

are people who have the problem able to build pico-examples?

Note that the workaround suggested is *not* the correct fix as it will break other compiler versions (that is what the version check is there for in the first place)

kilograham, 2021-11-25:
Also:

> In file included from /usr/include/newlib/stdio.h:35:0,
                 from /home/pi/pico/RPI-Pico-Cam/tflmicro/src/tensorflow/lite/micro/rp2/debug_log.cpp:40:

Including the error prone line in debug_log.cpp would be helpful.

lurch, 2021-11-25:
> > In file included from /usr/include/newlib/stdio.h:35:0,
from /home/pi/pico/RPI-Pico-Cam/tflmicro/src/tensorflow/lite/micro/rp2/debug_log.cpp:40:

> Including the error prone line in debug_log.cpp would be helpful.

Looks like that's https://github.com/ArduCAM/RPI-Pico-Cam/blob/master/tflmicro/src/tensorflow/lite/micro/rp2/debug_log.cpp#L40 ?

Hermann-SW, 2021-11-25:
Nothing Arducam specific, it is just "#include <stdio.h>" ;-)
https://github.com/ArduCAM/RPI-Pico-Cam/blob/master/tflmicro/src/tensorflow/lite/micro/rp2/debug_log.cpp#L40

kilograham, 2021-11-27:
> are people who have the problem able to build pico-examples?

^ this

ChrisLawther, 2021-11-28:
I can build pico-examples, but am also suffering the same problem with Pico SDK 1.3.0. A fairly minimal example to trigger it is attached.

Switching between `1.2.0` and `1.3.0` by `git checkout $tag` in $PICO_SDK_PATH toggles the undesirable behaviour.

Renaming `main.cpp` to `main.c` and making the corresponding edit in `CMakeLists.txt` does not cause the problem with either version.

[Minimal.zip](https://github.com/raspberrypi/pico-sdk/files/7613614/Minimal.zip)

lurch, 2021-11-28:
Thanks @ChrisLawther :+1: 

@kilograham Using this new example, this sequence below fails in a `debian:10` docker image, but actually builds okay in a `debian:11` docker image (again, with Ubuntu x64 being the OS that I'm running Docker on)
```
apt update && apt install -y git python3 g++ cmake gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib wget unzip
git clone https://github.com/raspberrypi/pico-sdk.git
wget https://github.com/raspberrypi/pico-sdk/files/7613614/Minimal.zip
unzip Minimal.zip && cd Minimal
mkdir build && cd build
PICO_SDK_PATH=../../pico-sdk cmake ..
make
```

(and as Chris suggests, if I instead do `git clone https://github.com/raspberrypi/pico-sdk.git -b 1.2.0` then it builds on `debian:10` as well; so I guess this counts as a regression? If the problem is indeed with C++ vs C, perhaps it's worth adding at least one pico-example that uses C++ ?)

kilograham, 2021-11-28:
ah `.cpp` is a new and good data point... yes we should add a cpp file to our SDK tests (I do test .cpp frequently in other things, but not across all compiler versions)

Martin-HiPi, 2021-11-30:
Affected as well, building pico-extras directly on a Raspberry Pi 4 using Buster (up-to-date):
arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0

Example:
```
In file included from /usr/include/newlib/stdio.h:35:0,
                 from /usr/include/newlib/c++/7.3.1/cstdio:42,
                 from /home/pi/projects/Pi-Pico/pico-extras/test/sample_conversion_test/sample_conversion_test.cpp:7:
/home/pi/projects/Pi-Pico/pico-sdk/src/rp2_common/pico_platform/include/pico/platform.h:237:31: error: duplicate 'inline'
 #define __force_inline inline __always_inline
                               ^
```

kilograham, 2021-12-13:
merged into develop for 1.3.1

closed 2021-12-13 by kilograham