raspberrypi/pico-sdk #662: PICO_SDK_PATH must be specified for flash boot

malek-luky, 2021-11-16:
I have trouble running $make V=1 (the third step in Installation from README). I am using Ubuntu LTS 20.04 WSL2 on Windows 10. I would be grateful for any help.

[image]

Thank you very much in advance.

lurch, 2021-11-16:
I don't believe that error message isn't coming from `pico-sdk` - perhaps it's a problem with "nuttx" ? :man_shrugging:

aallan, 2021-11-17:
If you're installing the SDK on Windows you might want to look at this [third-party installer](https://github.com/ndabas/pico-setup-windows) which simplifies things quite a bit.

malek-luky, 2021-11-17:
Thank you both for your comments. I am installing the SDK on WSL2, and also building and compiling the code there. I have found the line of code where the problem happens.

[image]

Unfortunately, I still can't figure out the main reason, because the PICO_SDK_PATH should be correct. A picture from the same terminal window is below.

[image]

lurch, 2021-11-17:
Again, that's code that comes from an external repo, so you'll need to open an issue there.

aallan, 2021-11-17:
Oh. Sorry, but there is no official support for installing the SDK under Windows Subsystem for Linux at this time. Our official installation route for users running Windows is using the Windows native instructions in Chapter 9 of the [Getting Started](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) book. This is the method the [third-party installer](https://github.com/ndabas/pico-setup-windows) I mentioned above takes.

closed 2021-11-17 by aallan