raspberrypi/pico-sdk #663: three USB CDC stop responding and connecting after a couple minutes

nikisalli, 2021-11-18:
pico-sdk version: 1.3.0
tinyUSB version: 0.12.0

I started by using [Noltari's pico-uart-bridge example](https://github.com/Noltari/pico-uart-bridge) as a base for my project and it works just fine under every load and condition but I observed that by simply adding a third cdc usb device and simple i/o operations like adc readings and/or gpio handling in the main loop, soon or later, tinyUSB completely stops responding to any connection (with an inuput/output error on linux) and stops sending or receiving any packet until reboot.
it is indefinitely stable if nothing is sent to the usb devices but it crashes as soon as, for example, one of the ports is maxed (cat /dev/urandom > /dev/ttyACM0) and another port is under any load condition (10 bytes every 100ms for example)

## REPRODUCIBLE TEST
[this crashes after a couple minutes](https://gist.github.com/nikisalli/233e90f52fbe470be6f9d7157120b476) with full load on one of the devices and little load on another (10 bytes from urandom every 100ms is what I tested with)
what I changed from Noltari's project:
- added third CDC device
- changed buffer sizes
- added example gpio and adc interactions

notice that three CDC uarts work just fine while not using any interrupt / spi / gpio / adc interaction. (tested using Harrywalsh's 3 cdc device adaptation https://github.com/harrywalsh/pico-hw_and_pio-uart-gridge of the original project from Noltari)

## DETAILED TESTS
What I tried so far:
- increasing CFG_TUD_CDC_EP_BUFSIZE to 256 and 512
- increasing CFG_TUD_CDC_RX_BUFSIZE and CFG_TUD_CDC_TX_BUFSIZE to 256 and 512
- trying various IN/OUT EP addresses
with the ones used now being:
```
#define USBD_CDC_0_EP_CMD 0x81
#define USBD_CDC_0_EP_OUT 0x02
#define USBD_CDC_0_EP_IN 0x82

#define USBD_CDC_1_EP_CMD 0x83
#define USBD_CDC_1_EP_OUT 0x04
#define USBD_CDC_1_EP_IN 0x84

#define USBD_CDC_2_EP_CMD 0x85
#define USBD_CDC_2_EP_OUT 0x06
#define USBD_CDC_2_EP_IN 0x86
```
- in my original firmware I need to make some spi transactions, read some gpios a couple time in a time critical code section and read some adcs, everything at a 50Hz rate. this is of course easily manageable for the pico but I did some long stability tests and observed that:
	- adc + gpios + spi readings -> tinyusb crash after a couple seconds
	- adc + spi -> tinyusb crash after a couple minutes
	- adc + gpios -> tinyusb crash after many hours
	- adc only -> no detectable crash after 3 hours of continuous flood on two of the usb devices
	- gpio only -> it crashed once after 2 hours of continuous testing
("adc" should be interpreted as reading one or more adc inputs at 50Hz , "gpios" as a fast burst of 50 gpio toggles every 20ms (50Hz), "spi" as reading an external interrupt driven IMU)
- handling uarts on core0 and core1
	- uarts handled on core0 -> shorter time until crash
	- uarts handled on core1 -> much more stable
- not using Noltari's FIFO buffer -> crashes as well. this proves that it is not the fifo making it crash (I tried  using a premade and thoroughly tested fifo and tinyusb freezes anyways)
- removing any interrupt related code -> didn't solve the problem

## strange behavior I noticed
garbage data is continuously being sent from the host to the third usb device but I couldn't determine if it is a firmware problem so this observation shouldn't be taken too seriously

as soon as the adc / gpio / irq / spi part is removed everything works just fine even with huge data loads

stepping through GDB didn't show anything unusual to my inexperienced eyes

running the firmware from the external flash instead of loading it in ram from gdb using picoprobe showed no different behavior

lurch, 2021-11-18:
> tinyUSB completely stops responding to any connection (with an inuput/output error on linux)

> as soon as the adc / gpio / irq / spi part is removed everything works just fine even with huge data loads

Apologies for the naive questions (I don't know much about USB) but is it "tinyUSB" that falls over, or the "Linux USB driver" ? Perhaps the Linux USB driver is expecting a response to USB packets within a certain timeframe, and if the Pico is "busy" running adc/gpio/spi tasks, then it isn't able to respond to all the USB requests fast enough; and so the Linux-side of the USB connection assumes that the Pico has stopped responding, and so doesn't send it any further USB packets? :shrug:

nikisalli, 2021-11-18:
I think the linux usb driver isn't the culprit (or the main culprit) here and I'll explain why
- even when running the usb stack on its own core it hangs, even with very small interruptions to update a buffer
- I wasn't able to reproduce the issue with a simple delay in place of the hw interactions
- linux isn't complaining about the device, it sits there waiting for packets until the connection is artificially closed, at that point trying to open another connection results in input/output error. there are no warnings or errors in dmesg

lurch, 2021-11-18:
Fair enough, perhaps it's a nested-interrupts problem or something on the Pico causing things to go wrong in tinyUSB? :shrug: 
I'll back off now (apologies for the noise!) and let the experts (Liam / Graham / @hathach ) deal with it :slightly_smiling_face:

nikisalli, 2021-11-19:
update:
it actually doesn't matter how many cdc devices are declared, it crashes even with just one device.
the test I ran included: reading 3 adcs, read and write some gpios and do some spi transactions and a very short interrupt callback, everyone of this in sequence at 50Hz. I handle tud_task() on core1 and fill the buffers on core0. It got completely stuck after 15 minutes.
here's the commit for the full code I used for this test in particular https://github.com/nikisalli/nightmare-firmware/commit/4033acd563a5886c5e45bfa3003d04ac78a15682

nikisalli, 2021-11-21:
ok I did some tests and I have narrowed down the problem.
Apparently an hardware usb to serial converter chip has the ability to limit the throughput of the host to the specific set baud rate. For example a CP2104 chip lets linux copy only 115200bit/s if the baud rate is set to 115200bit/s.
This is not the case with the tinyusb stack, in fact I was able to copy Megabytes of data every second thus making the rp2040 instantly crash.
Forget everything said above as those tests were in fact only testing how much additional stress the rp2040 could handle before crashing due to the huge amount of data.
I think this is issue has more to do with the usb stack than pico-sdk so I'm closing this issue and opening a new one on tinyusb.

closed 2021-11-21 by nikisalli

lurch, 2021-11-22:
> I think this is issue has more to do with the usb stack than pico-sdk so I'm closing this issue and opening a new one on tinyusb.

Which appears to be https://github.com/hathach/tinyusb/issues/1205