raspberrypi/pico-sdk #667: Alternative PICO_BOARD configuration not working

malek-luky, 2021-11-24:
I am using Qt Py RP2040, and lastly, I was struggling with the pinout configuration. I used export PICO_BOARD="adafruit_qtpy_rp2040" to change the pinouts. During the compilation, it seems to use the correct pinouts.

```
Using PICO_SDK_PATH from environment ('/mnt/c/Users/Luky/Documents/Pico/pico-sdk')
PICO_SDK_PATH is /mnt/c/Users/Luky/Documents/Pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Using PICO_BOARD from environment ('adafruit_qtpy_rp2040')
Using board configuration from /mnt/c/Users/Luky/Documents/Pico/pico-sdk/src/boards/include/boards/adafruit_qtpy_rp2040.h
```

But defining my I2C port as i2c0, the pinout configuration remains the same as in `pico.h`. When I want to connect to SDA and SCL, instead of using GPIO 24 and 25, it's working only for GPIO 4 and 5. Am I doing something wrong?

lurch, 2021-11-24:
Your output above looks OK (although note that we _officially_ don't support WSL), and https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/adafruit_qtpy_rp2040.h#L48 suggests that it ought to be using GPIOs 24 and 25 for I2C.
Can you confirm that you're using the latest release (1.3.0) of the SDK? Are you using the `PICO_DEFAULT_I2C_SDA_PIN` and `PICO_DEFAULT_I2C_SCL_PIN` defines?

malek-luky, 2021-11-25:
Thank you, Lurch, for your help. I was compiling the wrong file, where I was not using the `PICO_DEFAULT_I2C_SDA_PIN` and `PICO_DEFAULT_I2C_SCL_PIN` exactly as you mentioned. I close this issue since it is solved.

closed 2021-11-25 by malek-luky