raspberrypi/pico-sdk #668: Computed goto in a _time_critical_func still points to flash memory

MikeDB1, 2021-11-27:
If I use a gcc computed goto inside a function prefixed with  _time_critical_func then the base index address for the jump table declated with .word at the end of the function still points to an address 0x1000xxxx, which is in flash, rather than an address 0x2000xxxx in RAM.  

I believe _time_critical_func is a RP2040 extension so it needs to look for and change this base index address.

kilograham, 2021-11-27:
`__time_critical_func` is just a macro which sets a section attribute. it has no control over where GCC puts static data. Yes I agree it would be nice if GCC put such auxiliary stuff in the same section as the function, but it doesn't

closed 2021-11-27 by kilograham

kilograham, 2021-11-27:
i guess you can work around by copying the addresses to RAM yourself

MikeDB1, 2021-11-27:
I haven't tried it but does __time_critical_func also get long jumps incorrect then as these are absolute as well ?   

If __time_critical_func can't deal with non position-independent code it really should say so somewhere.

MikeDB1, 2021-11-27:
> i guess you can work around by copying the addresses to RAM yourself

How could I know where they are going to be ?  They will move with every edit.

MikeDB1, 2021-11-27:
> > i guess you can work around by copying the addresses to RAM yourself
> 
> How could I know where they are going to be ? They will move with every edit.

Okay scrub that.  By declaring the jumptable as a global pointer and then initialising every element once inside the function before it enters the while(forever) loop it assigns the addresses in RAM, not those in the Flash the code was copied from by _time_critical_func

Messy but it seems to work.