raspberrypi/pico-sdk #669: Strange error messages compiling a cpp file

pschatzmann, 2021-11-30:
I am trying to compile one of my projects and I am getting the following error messages. Does anybody know what the problem could be and how to resolve it ?
```
[  5%] Building CXX object CMakeFiles/rp2040-i2s.dir/src/pico_audio/audio.cpp.obj
/tmp/ccTeVq1B.s: Assembler messages:
/tmp/ccTeVq1B.s:29: Error: selected processor does not support requested special purpose register -- `mrs ip,PRIMASK'
/tmp/ccTeVq1B.s:32: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/ccTeVq1B.s:42: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccTeVq1B.s:52: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccTeVq1B.s:60: Error: selected processor does not support requested special purpose register -- `msr PRIMASK,ip'
/tmp/ccTeVq1B.s:63: Error: selected processor does not support `sev' in ARM mode
/tmp/ccTeVq1B.s:85: Error: selected processor does not support requested special purpose register -- `mrs ip,PRIMASK'
/tmp/ccTeVq1B.s:88: Error: selected processor does not support `cpsid i' in ARM mode
```
I am having this problems in OS/X so I tried to compile on a Raspberry PI but I am getting the same errors.

kilograham, 2021-11-30:
There is something weird about your setup.

What does your `CMakeLists.txt` look like? what is the output when you run `cmake`?

pschatzmann, 2021-11-30:
That's what I thought as well but I did not chatch any mistakes. I try to generate a library:
```
cmake_minimum_required(VERSION 3.13)

set(CMAKE_CXX_FLAGS "-fpermissive")
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)
if (PICO_SDK_VERSION_STRING VERSION_LESS "1.3.0")
    message(FATAL_ERROR "Raspberry Pi Pico SDK version 1.3.0 (or later) required. Your version is ${PICO_SDK_VERSION_STRING}")
endif()

project(rp2040-i2s)
pico_sdk_init()

## fetch arduino headers
#include(FetchContent)
#FetchContent_Declare(arduino GIT_REPOSITORY https://github.com/pschatzmann/pico-arduino.git GIT_TAG main)
#FetchContent_Populate(arduino)

add_library(rp2040-i2s 
    ${CMAKE_CURRENT_LIST_DIR}/src/I2S.cpp
    ${CMAKE_CURRENT_LIST_DIR}/src/pico_audio_i2s/audio_i2s.c 
    ${CMAKE_CURRENT_LIST_DIR}/src/pico_util_buffer/buffer.c
    ${CMAKE_CURRENT_LIST_DIR}/src/pico_audio/audio.cpp
#    src/pico_audio/audio_utils.S 
    $<$<NOT:$<BOOL:${PICO_NO_HARDWARE}>>:${CMAKE_CURRENT_LIST_DIR}/src/pico_audio/audio_utils.S>

)

## We need the arduino headers to be able to compile
target_include_directories(rp2040-i2s 
  PUBLIC
    "src"
  PRIVATE
    "src/arduino-ext"
)

target_link_libraries(rp2040-i2s 
  pico_sync
  hardware_dma
  hardware_pio
  hardware_irq
  hardware_clocks
)
```

pschatzmann, 2021-11-30:
Found it the set(CMAKE_CXX_FLAGS "-fpermissive") created the troubles!

closed 2021-11-30 by pschatzmann

pschatzmann, 2022-08-20:
https://github.com/pschatzmann/arduino-audio-tools/wiki/Encoding-and-Decoding-of-Audio