raspberrypi/pico-sdk #677: Pico firmware self-corruption from power cycles

spit4520, 2021-12-07:
Howdy,

I am trying to bring the RP2040 into production for our industrial product. We have noticed a few odd issues, namely with the reliability of the QSPI flash on board the Pico. Sometime the board will not function normally from one power cycle to the next, and the only solution is to reflash the hardware.

I don't quite believe it to be an application issue in our software. These are trivial things, where we have 2 pins defined to drive relays high. Once we receive a message over I2C, parse it, clear the interrupt, and handle it in our main loop. The pins are supposed to be driven high or low based on the message. One of our relays functions just fine, but the other relay will not respond. We are using isolation BJT's in the hardware to isolate out the gate driving itself to prevent overloading the RP2040's pins.

Once we reflash the hardware with the same binary, everything will begin to work again. Are there any potential issues with the Pico during boot up, when its building a ROM cache from the QSPI flash? Once one of these issues appears, it permanently is present, you can only remedy this issue with a HARD reflash. We are unable to determine if the issue happens while sitting (a few weeks sitting in a production assembly) or on power up/power down (both of which involve removing power without warning). We have also noticed that alternating between a version of the firmware with debugging symbols (enabling stdout to usb) and the production version will sometimes remedy the issue. This is super inconsistent and has only started to display itself as we are at the tail-end of our development cycle. 

Are there any flags we can set in the build environment to completely overwrite the program storage? Or are there any issues we should be facing regarding this? We don't write to the local storage for anything currently, so there should be NO reason for the behavior to permanently alter itself in between boots.

kilograham, 2021-12-07:
You say "Pico" and "rp2040"; are you using a "Pico" or a bare RP2040. If the latter, which flash.

Note you can use `picotool verify` to check the flash contents against your binary when the device is in BOOTSEL mode.

Note you also say "reflash"; how are you reflashing? UF2 via BOOTSEL mode or over SWD?

lurch, 2021-12-07:
> One of our relays functions just fine, but the other relay will not respond.

Surely if it was "firmware corruption" (which seems unlikely) then _both_ relays would fail to respond, rather than just a single relay failing to respond? Could the relays switching on and off cause dips / brown-outs in the power-supply to the RP2040? Have you investigated making use of the RP2040's watchdog feature?

> We are using isolation BJT's in the hardware to isolate out the gate driving itself to prevent overloading the RP2040's pins.

Don't forget that the RP2040's pins only operate at 3.3V, connecting them to 5V could damage them - that would seem (to me) to be the most likely explanation for one pin continuing to work fine, whilst another pin works erratically? Although of course that doesn't explain why a "reflash" would "fix" things.

ronaldDMAS, 2021-12-26:
see: https://forum.arduino.cc/t/arduino-feather-rp2040-earle-philhower-howto-and-learnings/938404?u=ronaldgadget

I was convinced I had flash corruption untill I understood what was happening...

lurch, 2021-12-26:
@ronaldDMAS "Arduino", "Philhowers" and "PlatformIO" are all 3rd-party software projects, and not directly supported by Raspberry Pi. Any problems need to be reported to the respective projects, rather than `pico-sdk`.

closed 2021-12-26 by kilograham

spit4520, 2021-12-29:
Hi, I didn't realize I have email alerts off.

1.) @kilograham  (nice name lol) We are using the PICO until we can source cut tape for the RP2040, we need ~3,000 of them and its basically impossible to find right now. (If you know where I can buy the cut tape, I will HAPPILY stop buying pico's on a reel:D ) That being said we may be asking too much of the buck boost at times, but we have a second dedicated 24v -> 5v buck that feeds into the main VUSB lines on the PICO, so the supply should be relatively stable? We have also experience some of our hardware dying inbetween power cycles as in taking other IC's with it. We use an NFC model from NXP, and its proven to be QUITE problematic during power cycles.

2.) @lurch We are only utilizing 3.3v logic everywhere, any time we need to "touch" or interact with any real peripherals, we are utilizing a BJT as our dumb switch to trigger a relay. We are doing this to keep the high power (like 60v) PSU completely decouples from the lower power electronics, we don't want any noise or instability created amongst the ground planes. Since the Pico and the relay are powered off of 5V, the buck we have on board should provide plenty-o-power (3A)

Here is what I suspect to be the suspect line of code:

`
#define OUT GPIO_OUT
#define IN GPIO_IN

#define EXT_LED PICO_DEFAULT_LED_PIN
#define CHARGE 21
#define UNLOCK 20
#define LOCK_SWITCH 19
#define zeroPlace 12
#define onesPlace 11
#define twosPlace 10

#define SPK 18

const uint pins[] = {EXT_LED, CHARGE, UNLOCK, LOCK_SWITCH, zeroPlace, onesPlace, twosPlace, SPK};
const uint pin_dir[] = {OUT, OUT, OUT, IN, IN, IN, IN, OUT};

        for(int i = 0; i < (sizeof(pins)/sizeof(pins[0])) - 1; i++) {

                printf("Setting pin: %d to: %d\n", pins[i], pin_dir[i]);
                gpio_init(pins[i]); 
                gpio_set_dir(pins[i], pin_dir[i]);
        }

`

The above snippet should run the same every time, the memory tables shouldn't be moving, and no wacky shit with A/B firmware is being done currently. I realize that its not the same as static declarations, this is just simpler when you have a lot of pins.

3.) @ronaldDMAS and @lurch We use only the raw Pico-SDK, we don't use any abstraction layers beyond that. We also try to use dedicated types like `uint32_t` etc when possible. If its a simple lazy loop, then its not really needed, so I don't believe we are running into memory corruptions currently?

lurch, 2021-12-29:
I'm only a software-guy rather than a hardware-guy, but this seems to sound suspiciously like an [EMI](https://en.wikipedia.org/wiki/Electromagnetic_interference) problem to me? (Especially as you mention other ICs are failing too.) Are you able to provide additional shielding around the Pico board to try and isolate it from any electrical noise that might be coming from your 60V circuitry? :shrug:

aallan, 2021-12-30:
> We are using the PICO until we can source cut tape for the RP2040, we need ~3,000 of them and its basically impossible to find right now. (If you know where I can buy the cut tape, I will HAPPILY stop buying pico's on a reel:D )

This sounds like you should be talking to the Commercial Team at Raspberry Pi directly, if you're not already doing so?

MikeBuffham, 2021-12-30:
@spit4520 if you can provide me with a contact (email or phone) I will get you through to someone who can supply RP2040 on 500 pcs or 3.4k pcs reels

spit4520, 2022-02-15:
Hi everyone,

We resolved most of our issues. They mostly did come down to EMI issues and some software unhappiness. Our culprit was a dirty buck way up network off of our main board and connected elsewhere on our wireharness. It was causing some very unhappy ripples, but only in specific conditions and it would cause the boot cycle of the 2040 to be disrupted. I don't have a great answer on how, but the QSPI flash would become corrupted and need to be written completely over for any kind of a rescue to happen.

@MikeBuffham my email address is: scott@kuhmute.com I am one of the co-founders and its CTO. We really love this little chip and I personally want to embed it everywhere I can!

Thanks everyone, Cheers!

lurch, 2022-02-15:
@spit4520 See also https://www.raspberrypi.com/news/raspberry-pi-direct-buy-rp2040-in-bulk-from-just-0-70/ :smiley: