raspberrypi/pico-sdk #678: Is there a way to store a secret key with the Pico-SDK?

spit4520, 2021-12-07:
Howdy,

I know that the QSPI flash itself is located outside of the RP2040 which introduces security concerns. I am trying to find a way to take the RP2040 into a safe production environment. I would like to store a certificate/public key somewhere on the device that I can use to authenticate the device during boot up with the rest of our PKI. The fact you can dump the QSPI is disheartening, but I have seen that the pico-sdk was used to write the actual pico-bootloader. The bootloader is said to be set in the chip at silicon time, but I was wondering if there was an exposed page table or small section of the ROM that could be used as a safe lockable "EEPROM" protected by a fuse that just hasn't been exposed through the SDK? Its a long shot, but I think it would be a needed ability in order to bring the RP2040 into IoT like environments

Thanks

kilograham, 2021-12-07:
nope. all RP2040 are identical and have no writable (even one time) storage.

closed 2021-12-07 by kilograham

spit4520, 2022-02-15:
@kilograham its probably out of the scope of RP2040, but is there a more canonical method I am not aware of where there is such a thing as a locked external flash chip? Maybe on that takes in an entropy source and internally can leave certain sectors locked until something like a TPM challenge is met in order to allow for mutation? I am aware that the literal instructions themselves are being piped out of this chip, and then interpreted. I am more interested in a way to "lock" the chip without killing the potential to use it as a File System in run time and prevent my whole product from having the circuit reverse engineered in a Chinese board house/the firmware dumped and flashed onto another 2040

lurch, 2022-02-15:
Sounds like a question for https://forums.raspberrypi.com/ ?