raspberrypi/pico-sdk #679: hardware-flash flash_range_erase documentation error

MattMills, 2021-12-08:
https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__flash.html#ga5b88461b126b1c226121146c87d5abe2
` count    Number of bytes to be erased. Must be a multiple of 4096 bytes (one sector).`

https://github.com/raspberrypi/pico-bootrom/blob/7a751f28436a8ee4849ac7799eb3aa1457e6f091/bootrom/program_flash_generic.c#L327
` // block_size must be a power of 2.`

count == 40960, which is a multiple of 4096 bytes, but not a power of 2 causes my RP2040 to hang.
stepping through the same range in count = 4096 works successfully.

MattMills, 2021-12-08:
Also, to clarify this testing was performed in PICO_NO_FLASH=1 ram only compilation mode.

MattMills, 2021-12-09:
After trying to test this further now that I have a working debugger I wasn't able to reproduce, so it must have been something else going on... But if you could clarify which piece of documentation is correct (power of 2 vs multiple of 4096) that'd be good.

test code:

```
#include "pico/stdlib.h"
#include <stdio.h>
#include "pico/stdio_uart.h"
#include "pico/time.h"
#include "hardware/flash.h"

int main(){
        stdio_uart_init();

        while(true){
                sleep_ms(1000);
                printf("flash start");
                flash_range_erase(0x0, 40960);
                printf("flash end");
        }
        return 0;
}
```

lurch, 2021-12-09:
>  But if you could clarify which piece of documentation is correct (power of 2 vs multiple of 4096) that'd be good.

You seem to be mixing up the `flash_range_erase` function in the [SDK](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_flash/include/hardware/flash.h#L53) with the similarly-named `flash_range_erase` function in the [bootrom](https://github.com/raspberrypi/pico-bootrom/blob/master/bootrom/program_flash_generic.h#L22). The [SDK function](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_flash/flash.c#L63) has two parameters, and [calls](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_flash/flash.c#L81) the [bootrom function](https://github.com/raspberrypi/pico-bootrom/blob/master/bootrom/program_flash_generic.c#L327) which has four parameters.

MattMills, 2021-12-09:
One calls the other.

lurch, 2021-12-09:
Yes, and the "power of 2" that you're referring to is the `block_size` parameter in the bootrom function. Which the SDK-function always populates with `FLASH_BLOCK_SIZE` which is indeed a [power of 2](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_flash/include/hardware/flash.h#L43).
The "multiple of 4096" applies to the `count` parameter in the sdk function.

MattMills, 2021-12-09:
Ah I see I transposed the two params of `flash_range_erase(uint32_t addr, size_t count, uint32_t block_size, uint8_t block_cmd)`

That said, some convenience functions for writing a new flash in non BOOTSEL mode without mucking directly with the flash wouldn't hurt.

closed 2021-12-09 by MattMills

lurch, 2021-12-09:
You might find https://github.com/rhulme/pico-flashloader interesting?

MattMills, 2021-12-09:
Yes that could be useful, I've been working on a runtime library that can be included in any non-tinyUSB app to create a USB mass storage interface that can flash like the bootloader as well as the normal STDIO USB interface and two more USB serial ports for forwarding uart0 and uart1 over here: https://github.com/2bn-dev/tinyusb_multitool/