raspberrypi/pico-sdk #682: non-blocking queue_try_remove calls are blocking?

MattMills, 2021-12-12:
Edit:

While I was writing this, I realized I was initializing the queue_t as a pointer instead of a value; going to submit it anyway so that it will show up in google if anyone has the same problem, and if you want to add a check that the queue or spinlock is valid anywhere in there. 

Feel free to close if you don't feel additional validation is appropriate.

Hello,

I don't really have much experience with spin locks, but I can't reconcile the behavior of queue_try_remove with the documentation:

```If the queue is not empty function will copy the removed value into the location provided and return immediately with true, otherwise the function will return immediately with false.```

```(gdb) bt
#0  spin_lock_blocking (lock=0x20041f00) at /opt/pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h:276
#1  queue_remove_internal (q=q@entry=0x0, data=data@entry=0x0, block=block@entry=false) at /opt/pico-sdk/src/common/pico_util/queue.c:64
#2  0x10001abc in queue_try_remove (q=q@entry=0x0, data=data@entry=0x0) at /opt/pico-sdk/src/common/pico_util/queue.c:102
#3  0x200003de in tumt_stdio_usb_out_chars () at /opt/tinyusb-multitool/src/tinyusb_multitool.c:88
#4  0x200004aa in tumt_periodic_task () at /opt/tinyusb-multitool/src/tinyusb_multitool.c:42
#5  <signal handler called>
#6  to_us_since_boot (t=...) at /opt/pico-sdk/src/common/pico_base/include/pico/types.h:48
#7  time_reached (t=...) at /opt/pico-sdk/src/rp2_common/hardware_timer/include/hardware/timer.h:115
#8  sleep_until (t=...) at /opt/pico-sdk/src/common/pico_time/time.c:361
#9  0x100010cc in sleep_us (us=<optimized out>) at /opt/pico-sdk/src/common/pico_time/include/pico/time.h:102
#10 0x10001106 in sleep_ms (ms=ms@entry=100) at /opt/pico-sdk/src/common/pico_time/time.c:393
#11 0x100003aa in main () at /opt/led-mcu-rp2040/heavy_lifting_test.c:33
(gdb) finish
Run till exit from #0  spin_lock_blocking (lock=0x20041f00) at /opt/pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h:276
```

I've been waiting for this spin_lock_blocking function to finish for several minutes and it never does, it just blocks... which seems expected by the name. Looking at the GDB output, I'm not totally certain the queue is valid

For more background, I'm using code similar to the default stdio_usb, but trying to split it into two processes and add a buffer/queue in the middle as I was getting deadlocks between the printf mutex and the stdio mutex when things were printing within the interrupt, which, since I also have a mass storage device within tinyUSB, currently happens frequently to debug what is happening.

daveythacher, 2022-08-12:
Should test lock and yield. This will cause issues for coroutines, RTOS tasks and multicore, if I am not mistaken.

daveythacher, 2022-08-12:
This is tricky though. This may get documentation update. Did it fail cause there was nothing in queue or because the lock wasn't free. If the lock is not free the state of the queue is unknown.

kilograham, 2022-08-12:
^ if you'll notice he edited a long time ago to say this was his bug.

Also the spin_lock is not a problem, and just protects the queue state. if the code must block, then `lock_internal_spin_unlock_with_wait` is called which works fine in irq/multicore environment by default, and is replaced to do something RTOS task sensible in FreeRTOS

closed 2022-08-12 by kilograham

daveythacher, 2022-08-13:
You cannot see the queue state without the spinlock. The queue state check is non-blocking. However, the spinlock grab is. This is not documented. I do not believe this should result in a change to the API. However, this should be documented as behavior of the API.

The API tries to simplify the interface. However, the API has an undocumented behavior which requires the programmer to be aware of an solve. The API here is a helper for certain things. Stating that this is multicore and ISR safe is correct. However, the blocking behavior here is not fully documented. To a knowledgeable programmer this should be apparent, however is missed by an inexperienced programmer. 

Part of the issue is the programmer is not responsible for locking. Therefore, the programmer must implement their own queue or wrap this queue in their own spinlock. Meaning the programmer would have to be responsible for the locking which is opposite the intention of API.

Again, a disclosure of this would remove this issue. The primary issue is the ambiguity of try/non-blocking. Simply stating that this can block, will make this a try only API for the queue state. It does not block on the queue but is potentially blocking. This is not handled by the API and the programmer is on their own. 

A knowledgeable/experienced programmer would not have this issue, I agree with that. This is marked as thread safe potentially and it is **not completely thread safe for real time**. I also agree that using this API for real time may be a huge mistake. Again, something a knowledgeable/experienced programmer would know. However, I am not sure this API should be written explicitly for them. (Again, this is my understand of the intention here, which models itself more like Linux/GNU.) I am asking for a short warning.

kilograham, 2022-08-13:
this and every other API which refers to "non-blocking" (which isn't in the name, but is intended to distinguish from the `_blocking` variants) is not necessarily "lock-free" (to which you refer). The documentation for the used spin-lock APIs *do* document that they are only to be used for short periods of time (and indeed the variants used here also disable IRQs). The expectation is (and unless the user deliberately uses the same spin-lock for another purpose and locks it) the actuality is, that the acquisition of the spin-lock will not block for more than a trivial number of cycles.

To the inexperienced programmer who you champion, the effect is exactly what they would expect... the non-blocking variants return an error almost immediately if the action cannot be performed.