raspberrypi/pico-sdk #685: std::cin doesn't work

Robotechnic, 2021-12-22:
I wrote this code to test communication over usb with my pico but it doesn't work each time.
When it work, it only work a couples of times until stop working.
I doesn't understand why, can you help me?
Here is my code
```cpp
#include "pico/stdlib.h"
#include  <iostream>

int main() {
	// init gpio
	gpio_init(PICO_DEFAULT_LED_PIN);
	gpio_set_dir(PICO_DEFAULT_LED_PIN, GPIO_OUT);

	// init usb
	stdio_usb_init();
	while (!stdio_usb_connected()) { // blink until usb is connected
		gpio_put(PICO_DEFAULT_LED_PIN, 1);
		sleep_ms(250);
		gpio_put(PICO_DEFAULT_LED_PIN, 0);
		sleep_ms(250);
	}

	std::cout<<"Send 1 to turn on, 0 to turn off, other number to exit"<<std::endl;
	int ledState = 0;
	while (ledState < 2 && ledState >= 0) {
		std::cin>> ledState;
		std::cout<<"ledState: "<<ledState<<std::endl;
		gpio_put(PICO_DEFAULT_LED_PIN, ledState);
	}

	std::cout<<"Good bye !"<<std::endl;
	return 0;
}
```
I use `screen` command to send and receive data
I'm on arch linux and I use the last sdk version
Thanks

lurch, 2022-01-12:
Can you give more detail about "it only work a couples of times until stop working" ? E.g. a copy'n'paste of your serial-terminal session. You might also want to try asking for help on the [forums](https://forums.raspberrypi.com/).

Robotechnic, 2022-01-12:
I mean that the program take the input, change the led state and display it on console and after 2 or 3 tries, it just stop working. The only thing that works is to reset the Pico.

If you whan here is an exemple:
```
Send 1 to turn on, 0 to turn off, other number to exit
1
ledState: 1 # here the led power on
0
ledState: 0 # here the let power off
1
ledState: 1 # here the led power on
0
# nothing appended
1
# nothing appended
...
```

kilograham, 2024-08-09:
i believe this is fixed