raspberrypi/pico-sdk #686: Build fails on MSYS2 because false path

jibaer-izw, 2021-12-23:
The generated makefile fails in creating the boot_stage2:

```
/C/msys64/usr/bin/python3.9: can't open file '/D/w2/pico/pico-examples/build/pico-sdk/src/rp2_common/boot_stage2/D:/w2/pico/pico-sdk/src/rp2_common/boot_stage2/pad_checksum': [Errno 2] No such file or directory
mingw32-make[2]: *** [pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default_padded_checksummed_asm.dir/build.make:73: pico-sdk/src/rp2_common/boot_stage2/bs2_default_padded_checksummed.S] Error 2
mingw32-make[1]: *** [CMakeFiles/Makefile2:4040: pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default_padded_checksummed_asm.dir/all] Error 2
mingw32-make: *** [Makefile:91: all] Error 2
```

In the Makefile ../.bs2_default_padded_checksummed_asm.dir/build.make the path to the python script does not follow the path convention for MSYS2:

```
in pico-examples\build\pico-sdk\src\rp2_common\boot_stage2\CMakeFiles\bs2_default_padded_checksummed_asm.dir\build.make
...
pico-sdk/src/rp2_common/boot_stage2/bs2_default_padded_checksummed.S: pico-sdk/src/rp2_common/boot_stage2/bs2_default.bin
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --blue --bold --progress-dir=/D/w2/pico/pico-examples/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_1) "Generating bs2_default_padded_checksummed.S"
	cd /D/w2/pico/pico-examples/build/pico-sdk/src/rp2_common/boot_stage2 && /C/msys64/usr/bin/python3.9.exe D:/w2/pico/pico-sdk/src/rp2_common/boot_stage2/pad_checksum -s 0xffffffff D:/w2/pico/pico-examples/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default.bin D:/w2/pico/pico-examples/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default_padded_checksummed.S
...
```
After replacing the start of the path to the python script from D:/... to /D/... it works.

Does anyone knows what goes wrong in the makefile generation?

lurch, 2022-01-12:
I'm afraid we don't officially support MSYS2, so you'll need to ask for help on the [forums](https://forums.raspberrypi.com/). Although if you do get it working, we'll gladly accept a Pull Request.

AXKuhta, 2023-01-13:
I have been able to build the blink example under MSYS2 but it required a different set of workarounds:

```
cd pico-examples
mkdir build
cd build
export PICO_SDK_PATH=../../pico-sdk

# This is required to fix undefined reference to `__gxx_personality_seh0' when building elf2uf2
# Appears to be a peculiarity with MSYS2's CC.exe
export CC=gcc
export CXX=g++

cmake ..

# Then absolute paths must be replaced with relative paths in generated/pico_base/pico/config_autogen.h
# The paths should look like that:
# #include "../../../../../pico-sdk/src/boards/include/boards/pico.h"
# #include "../../../../../pico-sdk/src/rp2_common/cmsis/include/cmsis/rename_exceptions.h"

# This should produce a working blink/blink.uf2
make blink -j4
```

Just sharing my experience.

bjanders, 2023-03-25:
I had this working previously, without having to do anything special. I had later updated msys2 and moved some directories around, and now I get this same issue, even when I put all directories back where I had them when it was working. I can't figure what I changed or how it was working previously.

bjanders, 2023-03-25:
I found the difference! Previously I was using a Windows "native" Python installation, and now I was using MSYS2 Python. It compiles by explicitly setting the path to correct Python binary, e.g.,
```
 cmake -G "MinGW Makefiles" -D Python3_EXECUTABLE=C:/Users/User/AppData/Local/Programs/Python/Python310/python.exe ..
```
Not sure why it previously automatically used the correct one.

closed 2023-05-26 by kilograham