raspberrypi/pico-sdk #687: multicore_lockout_start_blocking assertion failure

jonbro, 2021-12-24:
```
void draw_screen()
{
    multicore_lockout_victim_init();
    while(1)
    {
// in the actual application, I'm doing some i2c work here, but this code also fails.
        tight_loop_contents();
    }
}

int main()
{
    stdio_init_all();
    multicore_launch_core1(draw_screen);
    sleep_ms(100);
    multicore_lockout_start_blocking();
    multicore_lockout_end_blocking();
}
```

with the above code, I'm getting the following assertion failure:
```
assertion "us_since_boot <= INT64_MAX" failed: file "C:\Users\...\pico-sdk\src\common\pico_base\include/pico/types.h", line 63, function: update_us_since_boot
```

I'm on sdk 1.3, and using pico probe to debug... at some point, this code was running for me.

lurch, 2021-12-24:
Looking at https://github.com/raspberrypi/pico-sdk/commit/684986aae6674c3892cc0d85a5d4daa19b0eca4b that would seem to imply that your `at_the_end_of_time` is getting set to a value > `INT64_MAX` ? (i.e. the topmost bit is set).
What OS and compiler-version are you using?

jonbro, 2021-12-24:
I'm on windows, using `gcc for arm-none-eabi 10.2.1` / `GNU Arm Embedded toolchain 10 2020-q4-major`

jsphpl, 2022-07-11:
Same with _gcc version 10.3.1 20210621 (release) (GNU Arm Embedded Toolchain 10.3-2021.07)_ on _macOS 12.4 (M1)_. Does anybody know of a workaround?

dangiu, 2022-07-28:
Same on Ubuntu with arm-none-eabi-gcc (15:9-2019-q4-0ubuntu2) 9.2.1 20191025 (release)

kilograham, 2022-08-17:
fix is merged into develop

closed 2022-08-17 by kilograham

RaresAil, 2022-10-11:
I do get this issue with the latest SDK version, i am calling the block from core 1 and victim is core 0

lurch, 2022-10-11:
@RaresAil Probably best if you open a new issue, and include example code that reproduces the problem?

RaresAil, 2022-10-12:
Sure

PR #936 (2022-08-08): fix delayed_by_us and delayed_by_ms to not return times > at_the_end_…
fixes #687