raspberrypi/pico-sdk #688: Number of channels in the ADC mux

davidedelvento, 2021-12-27:
In https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_adc/include/hardware/adc.h#L23 and https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_adc/include/hardware/adc.h#L106 (and obviously the corresponding html and PDF documentation) it says that there are 5 channels to the ADC multiplexer, however only 4 possibilities are given (GPIO pins 26, 27, 28 and 29) and the last option is somewhat implicitly implied is for the internal temperature measurement.

Given that on pag 4 of https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf it says `23 GPIO are digital-only and 3 are ADC capable` and that Fig.2 of the same document does not show any GPIO pin 29, I suspect that the error is in `adc.h` and that the mux is a 4-channel one (more reasonable number than 5 for such a piece of hardware).

Either way, one of the two documents is incorrect and needs to be rectified, and in my opinion it can be more clearly stated that GPIO 29 is not user accessible and would measure the temperature (if that's correct).

aallan, 2021-12-27:
You're confusing the RP2040 chip which has 4 ADC channels, and an additional fifth which is connected to the internal temperature sensor, and the Pico, which exposes only 3 of the 4 available ADC channels to external pins. The "Pico SDK" is really the "RP2040 SDK" so talks about the more general case of the RP2040 rather than just the Pico.

lurch, 2021-12-27:
Page 7 of https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf explains that on the Pico board, the fourth RP2040 ADC channel on GPIO29 is used to measure VSYS/3.

davidedelvento, 2021-12-27:
Thanks to both. Perhaps a short text can be added to the relevant section to avoid confusion, especially since the name of the document is "Pico SDK" rather than "RP2040 SDK".

closed 2021-12-27 by aallan