raspberrypi/pico-sdk #694: pioasm: Bad python blocking irq instruction

jagerkin, 2022-01-03:
pioasm generated python uses wait instead of block for irq instructions, resulting in the error:

`TypeError: unsupported types for __and__: 'bound_method', 'int'`

Demonstration code:

```
$ cat test.asm
.program test
    irq wait 0

$ pioasm -o python test.asm
# -------------------------------------------------- #
# This file is autogenerated by pioasm; do not edit! #
# -------------------------------------------------- #

import rp2
from machine import Pin
# ---- #
# test #
# ---- #

@rp2.asm_pio()
def test():
    wrap_target()
    irq(wait, 0)                          # 0
    wrap()

```

smurfix, 2022-05-04:
Is there a reason why this isn't (fully) merged yet?
I just hit this problem …

lurch, 2022-05-04:
It's been fixed on the `develop` branch; we haven't yet merged it onto the `master` branch because we've not done a [release](https://github.com/raspberrypi/pico-sdk/releases) of `pico-sdk` for a while. But SDK 1.3.1 might be coming soon...

PR #695 (2022-01-08): fix pioasm python output (#694)
Fixes #694
Issue:
`    irq wait 0`
was being translated as:
`    irq(wait, 0)                          # 0`
wait is incorrect, it should be block. After change:
`    irq(block, 0)                         # 0`