raspberrypi/pico-sdk #697: documentation

nitrologic, 2022-01-03:
Reference to undocumented function pico_enable_stdio_usb that takes two parameters (TARGET ENABLE)

https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2_common/pico_stdio_usb/include/pico/stdio_usb.h#L16

lurch, 2022-01-03:
I'm not quite sure what you're asking for, but see "4.1. Serial input and output on Raspberry Pi Pico" in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

nitrologic, 2022-01-03:
This issue affects the helpfulness of c/c++sdk  4.4.20. pico_stdio_usb section in which the ENABLE parameter is omitted from docs generated from line highlighted.

lurch, 2022-01-12:
Thanks, see #705

kilograham, 2023-05-26:
this was fixed in 1.3.1

PR #705 (2022-01-19): Clarify header-comments about pico_enable_stdio_ CMake-functions
Fixes #697
closed 2023-05-26 by kilograham